Zendure/Zendure-HA #278: Vergleich App "Intelligenter Mode" vs. Integration Smart Mode

Kieft-C, 2025-05-19:
Gab es bereits einen Vergleich der beiden Modi's in seiner Effektivität bzw. Regelungsgeschwindigkeit?

Ich habe meine Integration nochmal komplett neu eingebunden und nun kann ich den smarten Abgleich Modus nutzen.

Zusatzfrage: Der Sensor wird mir zwar nicht korrekt angezeigt nach eintragung "Konfiguration ändern" steht immer nur der standard power sensor drin und nicht der zuletzt eingetragene bzw. bekommt man nirgends feedback welcehr sensor genutzt wird, vielleicht kann man dies ändern?

Nun läuft nach div. Sekunden warten der smarte Abgleich Modus in der Integration an und regelt entsprechend meines Shelly total power sensors.
Es sieht ganz okay aus, aber schwer zu vergleichen.

Hat jemand die Qualität der Regelung mal genauer verglichen zu der Regelgeschwindigkeit des Zendure App internen Mode?

MacdiverOne, 2025-05-19:
Mit "neu konfigurieren" wird der Sensor richtig angezeigt.

Die Regelung ist dem Zendure-CT / Zendure-Cluster mindestens ebenbürtig. Wenn ich mit meinen heute 13,5 kWh Eintrag und 2 Hyper (je 4xAB2000S) + 1 Hub1200 (2xAB1000) sehe, wieviel in Richtung Netzbetreiber "verschütt" ging (< 100 Wh mit manuell eingestelltem Offset von -10W). Da träumt Zendure nur von.

Kieft-C, 2025-05-19:
> Mit "neu konfigurieren" wird der Sensor richtig angezeigt.
>
> Die Regelung ist dem Zendure-CT / Zendure-Cluster mindestens ebenbürtig. Wenn ich mit meinen heute 13,5 kWh Eintrag und 2 Hyper (je [...]

Super, vielen Dank. Ich bin immer nur auf "konfigurieren" gegangen und dort war er nie zu sehen. Das hilft.

Den OffSet hattest du dir selbst ja gebaut und wird ggf. mit zukünftigen Updates kommen, wenn ichs richtig mitgelesen habe.

Kieft-C, 2025-05-21:
Hab jetzt mal 1-2 Tage den smartmatch modus laufen lassen.
Grundsätzlich läuft er bei sehr ähnlich gleichbleibenden Lasten sauberer als der zendure CT-Mode.

Nur bei Schlagartigen höheren Lastwechseln ist der Smartmatch modus langsamer als der zendure CT-Mode.
Kann man dies irgend wie verändern / verbessern? Die Abfragerate etwas erhöhen das er früher reagiert?

Ich nutze Shelly 3em pro der glaub ich alle 3sec sendet, sodass der smartmatch modus eigentlich alle 2,5sec oder 2sec es abfragen müsste, da sonst im "worstcase" mind. 6sec lang nix passiert plus dann die regelzeit von x sekunden auf die leistung x on top kommt.

harrymayr, 2025-05-21:
> Ich nutze Shelly 3em pro der glaub ich alle 3sec sendet, sodass der smartmatch modus eigentlich alle 2,5sec oder 2sec es abfragen müsste,

Der Sensor wird nicht abgefragt (gepollt), sondern es wird auf ein Event (Änderung des State) von HA reagiert. Aber es sind noch minimale Zeiten gesetzt, die bei 3 Sekunden zur Zeit liegen. Ich teste das mal mit einer kürzeren Zeit. (Kannst du auch, wenn die die Datei const.py änderst)
Es ist aber damit kein festes Zeitraster (alle.3 Sekunden oder so) sondern immer wenn der Sensor sich ändert, aber mindestens 3 (könnte geändert werden) Sekunden nach der letzten Aktualisierung.

Bimmerfriend, 2025-05-21:
Moin, es hört sich irgendwie so an, als würdet Ihr Euch mit einer ähnlichen Anlage wie meiner auseinandersetzen...

Ich habe zur Zeit 3 Hub2000 mit je einem AB2000 laufen. Eventuell kommt bald noch ein Hyper2000 mit 1 oder 2 kWh dazu. 

Was ich suche, ist eine smarte Regelung bezogen auf den Summenstrom der drei Phasen, da es bei mir eine Einspeisung auf drei Phasen gibt plus einem WR ohne Zendure. 
Wie kann ich das vernünftig mit einem Shelly 3EM pro kombinieren? Also in der Art "wenn Power negativ, dann auf Laden, wenn Power positiv dann einspeisen aus den Akkus in gleicher Höhe (smarte Regelung). Bisher sehe ich den Wald vor lauter Bäumen nicht mit der fireson Integration...

Kieft-C, 2025-05-22:
> Moin, es hört sich irgendwie so an, als würdet Ihr Euch mit einer ähnlichen Anlage wie meiner auseinandersetzen...
>
> Ich habe zur Zeit 3 Hub2000 mit je einem AB2000 laufen. Eventuell kommt bald noch [...]

Jaein, Also du kannst dir unabhängig ggf. eine Automation mit Wenn, dann Abhängigkeiten bauen, aber die Integration regelt ja nur Zendure Produkte. Und der Integration is es auch egal ob du den Shelly Power Sensor mehrfach vergibst.
Da musst dir selbst ein wenig ein klares Konzept zurecht legen, aber hier in dem GitHub Issue Bereich wirst keine klare Info bekommen dazu.

harrymayr, 2025-05-22:
> Was ich suche, ist eine smarte Regelung bezogen auf den Summenstrom der drei Phasen, da es bei mir eine Einspeisung auf drei Phasen gibt plus einem WR ohne Zendure.

Die Integration verteilt die benötigte Leisting auf die Cluster. In deinem Fall würdest du den erstem Hub einen Cluster mit 800W zuweisen und alle anderen den Cluster mit Hub 1.
Deinen anderen WR kannst du nicht direkt einbinden. Auch kannst du die Phase an den der WR nicht entlasten, es sei den, du hast an den Hubs an den anderen Phasen mehr Kapazität der Batterien. Die Verteilung erfolgt nach Kapazität und Ladestand.
Dein Shelly muss einfach die Leistung liefern, die du aus dem Netz beziehst bzw. einspeist.

Bimmerfriend, 2025-05-23:
Danke für Eure Kommentare! Es scheint, ich habe das Thema noch nicht durchdrungen... 
BT? MQTT? - nutze ich doch eigentlich Teich mit dieser Integration beides nicht? 
Cluster? - will bzw. kann ich ja nicht nutzen, da jeder Ze durch Hub aus separater Phase sitzt?
Fremdprodukte? - die will ich doch gar nicht regeln bzw. Integrieren? Oder meint Ihr den Shelly?

Fragen über Fragen, werde wohl nochmal weiterlesen und forschen.

Eigentlich hatte ich die Hoffnung, dass ich mir mit Automationen etwas basteln kann, was mir eine Individualisierung der Laden/Entladen-Vorgänge ermöglicht. Also pro Hub eine Möglichkeit habe, Smart oder Laden einzustellen... Aber da habe ich das Konzept wohl noch nicht verstanden.

Bimmerfriend, 2025-05-26:
Also, trotz Lesen und Denken geht es bei mir nicht weiter. Ich kann ja nun wirklich nichts weiter falsch machen, eigentlich... 
Angemeldet beim Account, Auslesen funktioniert auch. Shelly ist als Sensor im Manager angegeben.
Aber weder wird die aktuelle Leistung des Schelly angezeigt im Manager, noch erfolgt die Regelung darauf. Versuchsweise habe ich jetzt mal zwei Hubs zu einem Cluster in der Integration zusammengeschlagen, aber es geht trotzdem nichts. 

Muss ich in der App noch was ändern oder abmelden?

Bimmerfriend, 2025-05-26:
Und noch ein Kommentar... 

Hab gerade noch einmal im Wiki gestöbert und die Integration konfiguriert. 

Nun bin. Ich komplett verwirrt - warum fragt die Integration nach Wlan credentials?
Und warum BLE?
Und: Es gibt ja den Verweis auf lokales MQTT. Allerdings nutze ich EQX als MQTT Server... wie konfigurieren ich das? Und warum?

harrymayr, 2025-05-26:
> Muss ich in der App noch was ändern oder abmelden?

Sinnvoll ist es, erst mal alle Energiepläne zu deaktivieren.

- Welche Version verwendest du? Sinnvoll ist es die neueste 1.0.43 zu verwenden
- das lokaleMQTT nutzt Bluetooth, um eine Verbindung mit dem Device aufzubauen. Wird genutzt, um die Kommunikation nicht über die Cloud laufen zu lassen. Braucht man nicht zwingend --> **erst mal nicht verwenden**
- Die WLAN credentials werden gebraucht, um eine Lokale Verbindung aufzubauen, meiner Meinung nach werden die aber nicht gebraucht, wenn keine lokale Verbindung gewünscht wird. Die Integration verlangt sie trotzdem. Da muss @FireSon mal schauen, ob die wirklich zwingend benötigt werden. 
- Wie oben schon geschrieben, die einzelnen Hubs ein **einen** Cluster einfügen.
- Nach den Einstellungen am besten HA neu starten
- Sollte es nicht funktionieren, ein Debug-Log erzeugen, ggf. persönliche Dinge löschen und hier einstellen.

FireSon, 2025-05-26:
@Bimmerfriend Hoffentlich klappt es 😄 

Die wlan credentials werden nur verwended in combination mit Bluetooth. 
Es wird verwended fur:
1) schalten von Cloud auf Lokale MQTT und zuruck.
2) reset Hyper2000, nach +-2 tagen gibt es Probleme mit die MQTT kommunikation de Hyper2000, ueber Bluetooth kann das behoben werden.

FireSon, 2025-05-26:
Das stimmt!

Bimmerfriend, 2025-05-29:
Guten Morgen,

ES FUNKTIONIERT!!!

Die Einstellung auf einen (!) Cluster war wohl das Problem. 

Ich habe nur umgestellt auf "Cluster 2400" an einem Hub und die anderen beiden Hubs auf "Teil des Cluster 1". Danach lief es und die drei Akkus sehen jetzt recht dynamisch aus. Ich werde beobachten, wie es mit der Dynamik und dem Überschussverwalten läuft, bisher sieht alles gut aus.

Bleibt die Frage: Wenn ich jetzt zu den drei Hub2000 noch einen Hyper 2000 hinzufügen, kann ich dann auch noch AC Laden vom abgesetzten Wechselrichter realisieren?

Danke Euch nochmal für die tolle Integration (@FireSon ! :-) ) Und Eure Unterstüzung!

> Ok, danke für die klärenden Details: Neueste Version 1.0.43 ist installiert Lokales MQTT ist aus BLE habe ich nicht installiert
> 
> Noch nicht versucht: Alles über einen Clister laufen lassen, hatte bisher gedacht, es nutzt dann den ZenLink Kram...
> 
> Ich werde es probieren...

Bimmerfriend, 2025-05-30:
Soo, kleine Korrektur: ich hatte in der App wohl noch einige Zeitpläne aktiviert, die habe ich jetzt ersteinmal rausgesucht. In welchem Modus dann die drei Zendure laufen in der App ist egal, oder miss ich auf einen bestimmen Modus in der App stellen?

> Guten Morgen,
>
> ES FUNKTIONIERT!!!
>
> Die Einstellung auf einen (!) Cluster war wohl das Problem.
>
> Ich habe nur umgestellt auf "Cluster 2400" an einem Hub und die anderen beiden Hubs auf "Teil des [...]

harrymayr, 2025-05-30:
> Oder ist das doof gedacht?

Ist nicht doof gedacht. Das einzige was halt passieren könnte ist, dass nicht genug AC Power für den Hyper vorhanden ist. Aber wie du es beschreibst, scheint das kein Problem zu sein. 
Beim Entladen hast du halt an der Phase wo der Hyper dran hängt dann 2/4 der Leistung (ein Hub + der Hyper), an den anderen je 1/4.

harrymayr, 2025-05-31:
> Solarflow 2400 AC
Aber da bleibt dann die Frage, ob er integrierbar ist in diese Integration?

Ist Integrierbar, wie es aussieht aber nur mit dem Haupt-Zendure-Account, weil das Device Sharing damit wohl nicht geht.

> denn zur Zeit kann ich ja die Modi (Laden, Entladen, Aus, ...) von einzelnen Zendure Gerätennochnicht steuern...

Warum willst du das tun? Der Smart Mode macht das alles automatisch.

> damit die Regelung nicht zu hektisch wird? Oder ist das Zeitverhalten egal?

Die Regelung der Integration selbst erlaubt nur alle 3-5 Sekunden einen neuen Wert (theoretisch, praktisch sehe ich bei mir ca. alle 7 Sekunden einen neuen Wert)

> die Energie-Aufteilung beziehst Du auf die Regelungstheorie des Smatmodes der Integration? Oder auf meine Phasenverteilung?

Der P1 Sensor liefert einen Wert, der sagt, ob Strom geliefert oder gebraucht wird.
Diese Leistung wird auf alle Geräte, die einem Cluster zugeteilt sind, im Verhältnis der angeschlossenen Batterie-Leistungen und SOC aufgeteilt.
Die Integration weiß natürlich nicht, an welcher Phase die einzelnen Devices angeschlossen sind. In deinem Fall bei 3 Hub (jeder an einer anderen Phase) und einem Hyper (an einen der 3 Phasen) ergibt sich dadurch bei gleicher Batterie-Größe beim Entladen an einer Phase (Hub und Hyper) 2/4 der Leistung da die angeforderte Leistung eben gleichmäßig verteilt wird.
Bei Überschuss wird die Leistung bisher auch gleichmäßig verteilt. In deiner Situation mit 3 unterschiedlichen PV Feldern könnte es sein, das den beiden Hubs an dem einem Feld mehr Leistung zugewiesen wird, als PV Leistung am Feld anliegt.
Die gleichmäßige Verteilung führt z.Z. auch dazu, dass wenn man manuell den Hyper AC laden wollte, den Hubs auch ein Teil der freigegebenen Leistung zugewiesen wird, obwohl die gar kein AC Laden können. 
Da sind aber @FireSon und ich dran und werden das noch verbessern.

harrymayr, 2025-05-31:
Solarflow 2400 AC: Das ist es, was ich aus diversen Benutzerkommentaren rausgeht habe. Selbst nicht getestet.

Gerätesteuerung: Dein 3. Feld wird doch immer Überschuss erzeugen. Und denn kannst du ja für die Hubs nicht nutzen.

Regelung/Zeitverhalten: Ggf. wird in einer der nächsten Releases eine Funktion vorhanden sein, die dir einen Offset erlaubt, so dass du im Zweifel mehr ins Netz einspeist, als beziehst.
Die Hubs sind in der Regelung sehr langsam, gerade, wenn du größere Sprünge in der Leistungsabnahme hast, kann das dazu führen, dass du mehr Strom beziehst. Da bin ich auch dran am arbeiten, das in Kombination mit einem Hyper die langsame Regelung der Hubs ausgeglichen wird.

Bimmerfriend, 2025-05-31:
Klar, passt soweit. 

Aber zur Gerätesteuerung: das dritte Feld erzeugt bei schlechtem Wetter tw. Gerade genug, um den aktuellen  Bedarf zu decken. Also wäre dann die Prio: Feld 1 und 2 sollen die Akkus der Hubs laden, Feld drei den Momentanverbrauch decken. Wenn Überschuss aus 3, dann höhere Ladung an Akkus 1 und 2 für Beendigung des Lafens Hub Akkus. Erst dann soll bei Überschuss an AC Akku gehen. 

Bei super Wetter spielt das keine Rolle, da ist ab Mittag eh alles voll und es gibt genug Überschuss. Mir geht es nur um schlechteres Wetter bzw. Dann auch um Winterbedingung (Feld 1 und 2 sind anders ausgerichtet als Feld 3).

Ich hoffe das macht so Sinn, zumindest als Erklärung...

harrymayr, 2025-06-06:
Soweit ist das alles korrekt, die Geräte erscheinen auch immer (unabhängig von der Clustereinstellung) getrennt. Jetzt müssten die angeforderten Leistungen 50/50 auf die beiden Hyper verteilt werden.
Du hast nicht geschrieben, was nicht funktioniert.

klucon2, 2025-06-06:
Hi,

also Sensor ist folgendermaßen eingetragen:
[image]

Beim Sensor selber sieht es so aus:
[image]

In der Zendure App sind alle Modi aus.
Beide Hyper stehen auf "allow access energy export" und "allow cluster".
Output mode 0W
Alle FW up-to-date.

Smartmatching an und aus geschaltet. Leider ohne Veränderung :-(

Das Hyper Energy Programm zeigt folgendes:

[image]

Ich entferne jetzt mal die Integration und setzt sie neu auf. Melde mich dann nochmal.

klucon2, 2025-06-06:
Also, nach dem neuen Setup zeigt es mir jetzt nur 3 devices an (davor 6) und weniger entities.

[image]

Aber leider wird immer noch nicht eingespeist.
Hier auch nochmal das Energy Programm:

[image]

[image]

Habt ihr weiter Ideen woran es liegen könnte?

klucon2, 2025-06-06:
Anbei das log. 

[home-assistant_zendure_ha_2025-06-06T19-49-09.418Z.log](https://github.com/user-attachments/files/20633612/home-assistant_zendure_ha_2025-06-06T19-49-09.418Z.log)

Kieft-C, 2025-06-06:
Hast du nach dem deinstallieren und neu installieren HA komplett neugestartet, nicht nur die yaml-neuladen.
Dies solltest du machen, damit alle Geräte & Entitäten geladen werden.
Erst wenn dir alle Geräte richtig angezeigt werden, mit smart matching anfangen.

Wird dir bei den Akku's ein Füllstand in % angezeigt?

Was mich wundert in deinem Log File, dass "capacity 0" überall steht.
Also wie als wäre einfach dein Akku leer? Aber ist nur geraten, da stecke ich nur teilweise drin.

Aber generell lade einmal HA neu und checke das die Geräte alle Daten liefern die schlüssig sind und sich in der App auch zeigen.

klucon2, 2025-06-06:
Hallo Zusammen,

also hatte tatsächlich den Neustart vergessen. Nach dem Neustart funktioniert alles wie gewünscht 🤗
Vielen Dank für die Unterstützung!!