Zendure/Zendure-HA #282: Waste energy returned to grid with battery low ( 1.0.41-pre8 )

FrColin, 2025-05-20:
Today battery was low and fall to SocMin when a big consumer was required 1200W, 
the hyper goes to bypass since no more battery power .
but when the consumer get off , the hyper 2000 stay in bypass mode leaking solar power to the grid.
it should make the output to home  limited to the demand and charge the batterie with the surplus
you can see the issue in the following screen capture and log file

**at 13:03:06** we have capacity 0 : `Update setpoint: 1279 stateManagerState.DISCHARGING capacity: 0 max: 1200`
**at 16:34:01** I stop zendure manager : `Update operation: 0 from: 2` to restore functional system
**at 16:34:03** bypass turn off : `Update binairy_sensor: Hyper2000-pass => False`
 

[image]

[bug_inject.log](https://github.com/user-attachments/files/20351654/bug_inject.log)

hope the log can help

FrColin, 2025-05-21:
A details of values
bypass so input and ouput pack is 0 also grid input is 0 because no charging from grid
 
[image]

harrymayr, 2025-05-21:
I checked the data and have a proposal to change the code. But I can't test in on my own, because I don't have some panels direct on my Hyper. If you are able to change the file \config\custom_components\zendure_ha\zendurermanager.py, you can give it a try.

`                d.powerAct = d.asInt("packInputPower") - d.asInt("outputPackPower")`
`                if p.powerAct != 0:`
`                    d.powerAct += d.asInt("solarInputPower")`

Be sure to have the right indent. This change should allow the Hyper to start charging if enough power is available. It may be a good idea to have logging on.

[image]

The original code is this:

[image]

harrymayr, 2025-05-21:
FYI: there was a typo in the code and image. I corrected it.

harrymayr, 2025-05-22:
As long as powerAct is positive, the integration assume that the device is discharging. Online if powerAct is 0, it goes into Standby and then decide to charge or discharge depending on P1.

FrColin, 2025-05-25:
@FireSon , sorry it is not possible to test now because not at home .
I will reply later
Thanks

FrColin, 2025-06-09:
Hello, i am back ,
@harrymayr the code mod doesn't work it still reject energy to the grid with the battery at 14% 
the situation is not stable 
i join some logs 

[home-assistant_zendure_ha_2025-06-09T11-28-15.325Z.log](https://github.com/user-attachments/files/20655531/home-assistant_zendure_ha_2025-06-09T11-28-15.325Z.log)

[home-assistant_2025-06-09T11-22-00.999Z.log](https://github.com/user-attachments/files/20655511/home-assistant_2025-06-09T11-22-00.999Z.log)

[image]

[image]

[history_zendure_bug_1.48.mod.csv](https://github.com/user-attachments/files/20655510/history_zendure_bug_1.48.mod.csv)

harrymayr, 2025-06-09:
Is it possible, that the mqtt-logging is off? I'm missing a lot of data in the log.

FrColin, 2025-06-09:
> Is it possible, that the mqtt-logging is off? I'm missing a lot of data in the log.

yes it is off sorry , should i retry the operation with mqtt-logging on , with the "code mod" and without ?

harrymayr, 2025-06-09:
Hmm, does your Wifi has a special name, perhaps with characters used in French language? In the line with the error it should show up something like:
`/gDa3tb/Hyper 2000/properties/report => {'wifiName': 'NEXUS-M', 'mac': '94:c9:60:--:--:--', 'ip': '172.16.xyz.xyz', 'encryption': 'WPA2 AES', 'messageId': 2628675, 'product': 'solarFlowPro', 'timestamp': 1749287017, 'properties': {'packNum': 1}, 'packData': [{'sn': 'CO4H------------'}]}`

harrymayr, 2025-06-09:
> i check /gDa3tb/Hyper2000/properties/report and all is correct

Oh, there are many lines with this header, and they are all different and depend on the requests before.

And I wonder, that it tries to get some data for the same device inbetween 3 seconds.

[image]

FrColin, 2025-06-10:
In this log i do not make code change, to be sure that the situation still exist.
I will make the change tomorrow, my previous change was not very concluant but i will redo it with full mqttlog.

FrColin, 2025-06-11:
@harrymayr Hi,
today i tested with the code modification it seems ok , but since sun shine today I am not in the exact same situation bat level > 30%
I would just ask a question why to use the pack power to determine the actualPower, it should be easy to use the "output_home_power" and take in account the effective output power after DC/AC convertion 
some thing like:
`            powerActual = 0`
`            for d in ZendureDevice.devices:`
`                d.powerAct = d.asInt("outputHomePower")`
`                powerActual += d.powerAct`

this is the log of ma last test with code mod you preconise

[bug_pass_mode_withcodemod.log](https://github.com/user-attachments/files/20697058/bug_pass_mode_withcodemod.log)

hope this will help
thanks a lot for your  help

harrymayr, 2025-06-11:
> I would just ask a question why to use the pack power to determine the actualPower, it should be easy to use the "output_home_power" and take in account the effective output power after DC/AC convertion
> some thing like

This will not work, because the powerActual also control charging in conjunction with the P1.

FrColin, 2025-06-11:
But the hypercar automatically charges the battery if there's excess energy?
If solar production exceeds the house's demand, the battery charges.

harrymayr, 2025-06-11:
On the one hand, this section is for all devices, also devices which has no possibility to charge with AC or which have no PV input like the SolarFlow 2400AC. And this is the other point. I personally have no PV panels connected to my Hyper. I have a large (7.15kW peak) PV on my roof, build 10 years ago and had to change my inverter to add a battery on the DC side. So I decided to use a Hyper and charge AC (knowing that this has not the best efficency  ~80%). 
Be sure, that this calculation fit most of the needs (with the change you test, hopefully all needs :-D   )

harrymayr, 2025-06-11:
And the Hyper as also not the cool between 13:00 and 15:00  8-|

[image]

harrymayr, 2025-06-12:
> It may be necessary to add a fan to cool it further.

Yes, I think this is the reason, the power drops.

harrymayr, 2025-06-14:
Hi @FrColin:
I think, everything works now as I would expect.
Around 13:25 your battery is empty (at minSoc), but you have still a high need of your house, which stops a few minutes before 15:00. In this time, all the solar power goes into the house grid. Then it uses a few minutes all solar power to start charging the battery and around 15:00 the demand from your house is less than the PV power, and the rest of the power loads the battery.
Do you see or expect something different?

[image]

FrColin, 2025-06-15:
Hi,
I think this is working well now , 
We can close the issue now.

closed 2025-06-15 by FrColin

harrymayr, 2025-06-15:
The code changes will be in the next release.