Zendure/Zendure-HA #284: Mqtt

Spottttt, 2025-05-21:
@FireSon 
Bonjour

J'ai un HA en conteneur et mqtt dans un autre conteneur avec une IP différente.
Comment configurer l'intégration en Mqtt Local ??

Hello,

I have an HA in a container and MQTT in another container with a different IP address.
How do I configure the integration in MQTT Local?

sankukace, 2025-05-21:
Bonjour,

La demande est identique à ma demande je pense :) #254 

-----------

Hello,

I believe the request is the same as mine :) #254

FireSon, 2025-05-22:
Yes, I will address it in the next release. The integration will use the same broker + username and password as specified in the MQTT integration

FireSon, 2025-05-23:
Can you test version 41? Make sure that you have added the MQTT integration, it is used to create the connection!

[image]

closed 2025-05-25 by FireSon