Zendure/Zendure-HA #286: Infinite reboot

Zanooon97, 2025-05-21:
@FireSon Hello FireSon, i write to u because @MarkP12345 said you know this problem about the retained mqtt message an will fix it in the next update. How long will it take to make a fix for this?
The Zendure Support are not answesring on my ticket, I tried to call them today but never get a chance. 

Thank you for your work

This said MarkP:

I had the same problem with a SF 800 Pro. At least it sounds as an identical problem. Via Bleutooth I could operate the SF Pro 800, but not via Wi-Fi via the cloud. Exactly the things you describe.

This was the analysis of Zendure:
We have discovered that your device is stuck in an infinite reboot loop because it has been triggered by a registration response command. Did you send an MQTT retained message to the topic iot/register/replay? If so, please stop this operation.
According to the design logic of the Zen SDK framework:

When the device receives a message under the topic iot//register/replay, it writes the registration information and triggers a reboot.

If the message is set as Retained, the device will receive the retained message again upon reconnecting to MQTT after rebooting, triggering the registration process and rebooting again.

This causes the device to enter an infinite reboot loop.

Fireson is aware of this problem and will solve it in next update, he said. I think your only solution is to contact Zendure. They have fixed it for me in the end.

Zanooon97, 2025-05-21:
Can I send you a dip over paypal?

Zanooon97, 2025-05-21:
I send you a buyme a coffe Zanooon97

JaySNL, 2025-05-22:
Oh this seems identical to my issue right now as well!

Zanooon97, 2025-05-22:
make a warnung about this problem

FireSon, 2025-05-22:
@Zanooon97 Yes I know about it, this week there will be a new release which will fix this issue. The cause is a MQTT topic which keeps resending, and apparantly causes a reboot of the device. I will do it asap!

Zanooon97, 2025-05-22:
@FireSon Wil this Update fix my problem or is the only way the Zendure Support??

Wunni73, 2025-05-22:
[image]

[image]

MarkP12345, 2025-05-22:
LOL, you dont think we have tried this.....no this issue is more complicated than this.....

Zanooon97, 2025-05-22:
i tried to publish an empty retained message on thetopic iot/sn/register/replay 
and its not changing 

is this topic right ? 
@Markp12345

MarkP12345, 2025-05-22:
iot/......../......../register/replay

I dont know exactly what the names of these fields are, but the first one looks like the productcode in mqtt and the second one as an identifier of the product in mqtt (not the serial).

Zanooon97, 2025-05-22:
the first is the appkey and the second the devicekey, can you look in the log from HA and confirm this?

JaySNL, 2025-05-22:
> LOL, you dont think we have tried this.....no this issue is more complicated than this.....but thanks for the suggestion

Zendure can reset it, remotely. So it's something server related. What it is, i don't know, but they said "background processing to be done" before it 'worked again'.

The first code is the product identifier, the second is the model/type. The original IoBroker plugin does somewhat the same:
https://github.com/JaySNL/ioBroker.zendure-solarflow/blob/main/src/models/ISolarFlowDeviceDetails.ts

  productId?: number;
  productKey: string;
  productName: string;
  productType?: number;

i.e: https://github.com/JaySNL/ioBroker.zendure-solarflow/blob/2317bdf5521c415cf31ee778c9ab434e56c345f3/src/services/mqttService.ts#L34C1-L60C10

Zanooon97, 2025-05-22:
@JaySNL how can i get this codes?

MarkP12345, 2025-05-22:
You can find them in your mqtt explorer. The entry of the data starts in the topic with these codes.

JaySNL, 2025-05-22:
> You can find them in your mqtt explorer. The entry of the data starts in the topic with these codes. But I would try the support of zendure to fix it.

They fixed (both) solarflows on my side, but this does make me a bit weary to toggle things again before an actual fix/diagnosis is done that this cant happen again. They said to come with their own Home Assistant support though (official!).. which is very very very interesting.

MarkP12345, 2025-05-22:
> They fixed (both) solarflows on my side, but this does make me a bit weary to toggle things again before an actual fix/diagnosis is done that this cant happen again. They said to come with their own [...]

Yes I heard the same but in cooperation with fireson

JaySNL, 2025-05-22:
> what is the produktid for the sf800?

You're lucky i saved in nodeRed; SolarFlow 800 - /0/a4ss5P/tc56ygHY

Zanooon97, 2025-05-22:
I found in the logfile of HA this

[image]

JaySNL, 2025-05-22:
> I found in the logfile of HA this
> 
> [image]

Ngl, you dont need to blur out the id and productKey, only SN. ProductKey and DeviceKey pretty much universal.

Zanooon97, 2025-05-22:
its not working, zendure support only solution so😬😬😬

Zanooon97, 2025-05-22:
@MarkP12345 can you look in your logfile ( activate debug on zendur integration ) and look where the secend code comes from, i think devicekey. look for me pls,🙏😅

MarkP12345, 2025-05-22:
I cant help you, i dont use the integration anymore. To dangerous for the new devices, until fireson has solved it. I think it is good to contact zendure and let them solve your problem Thats the only way I am afraid

FireSon, 2025-05-22:
I have found the issue with the endless loop! In order to change the MQTT settings, the (older) device needed to reboot. In order to do that the replay topic was written with a reboot command. For each device this command was written to the topic iot/deviceType/deviceId/register/replay 

In order to 'remove' this topic, which would prevent the reboot, you have to write an empty payload to this topic. I have tested it, and my MQTT broker stopped sending the payload. I have added this to the startup routine. I am planning to do a new release asap, but I still need to fine tune some other communication issues.

Zanooon97, 2025-05-23:
Listen I have get the appkey and secret and mqtt url  from my windows cmd with the offical zendure github tutorial. 

with that I logged in in mqtt explorer but no topics are displayed.

I published an empty retained message to iot/prductk/devicek/register/replay. The productkey and devicekey i have found from the logfile in HA. 

But nothing changed.

Am I wrong with a setting?

@FireSon

Zanooon97, 2025-05-23:
Update @MarkP12345 

[image]

Zanooon97, 2025-05-23:
the support have solved my problem, have you another way to see in HA the devices? I need only the output home power for my calculation in HA, otherwise I have one sensor that is showing minus

JaySNL, 2025-05-23:
> the support have solved my problem, have you another way to see in HA the devices? I need only the output home power for my calculation in HA, otherwise I have one sensor that is showing minus

You could subscribe to the MQTT topic yourself through NodeRED or something and pull that relevant topic data into HA.

Zanooon97, 2025-05-23:
with this: https://github.com/Zendure/developer-device-data-report

Zanooon97, 2025-05-23:
@MarkP12345 Hei Mark you have a sf800 pro you said, do you have data in HA in a way?

JaySNL, 2025-05-23:
what do you need?

------------------------------------------------------------------------------------------
Met vriendelijke groeten / Kind Regards,

Jooshua Franssen

FireSon, 2025-05-23:
In release 41, I have added a fix to remove this topic form the MQTT server. Also the topic will not be written anymore. Hopefully this will fix the current problems. 
Maybe there is a brave owner of a SF800 who is willing to try 😥

Zanooon97, 2025-05-24:
hahah should I try, you have delete the part where the topic is writtn? for what was this topic good in the past?

Zanooon97, 2025-05-24:
should I update my SF800 to v.1.0.4 first?

MarkP12345, 2025-05-24:
Dare devil ;-) but it still works?

Zanooon97, 2025-05-24:
@FireSon can I try local mqtt now with my SF800?

[image]
[image]

FireSon, 2025-05-24:
@Zanooon97 Yes I think you can try, the 'reboot' topic will be removed as soon as the integration connects to your local MQTT. After it is connected, the SF800 will be redirected to the local MQTT so that should be OK...

FireSon, 2025-05-25:
I found out that the Hyper2000 actually writes this tag, in order to restart! In the last version >41 this problem should be resolved 😄

closed 2025-05-25 by FireSon