Zendure/Zendure-HA #287: Help enabling smart mode

HomeAssistantPim, 2025-05-23:
Finally I got my zendure batteries (2400AC).
Very happy, it was a long wait.
I've just setup the battery and also the integration and this worked instantly (many thnx!)
I got readings, even setup the dashboard with the new aggr sensors (thnx), some oddities and can't get smart mode to work for which I need some help.

The oddities:
charge rate limit of the 2400AC is 2400W, but the integration limits to 1200 (managed to fix this via the zendure app though)
sharing the device no longer seems possible as described in your documentation. The zendure app does not show an option at the end

The help:
I've setup the Zendure Manager with a HomeWizard P1 and set the manager to smart matching.
I have not setup HEMS.
The battery only has 2 modes in the integration AC in and AC out.
When I select either of them the rate goes to 2400W not matching my P1 readings.
Did I do something wrong or forgot something to make it more smart?

FireSon, 2025-05-23:
@HomeAssistantPim The integration adds the ZendureManager as a seperate 'device' which manages all devices for smart matching. In my case it looks like this:

[image]

Allthough the readme is a bit outdated the part about the ZendureManager is still correct.

Zanooon97, 2025-05-25:
is this working only with local mqtt? 
I have set my shelly pro 3 em as p1 meter sensor, but its not working I think. 

[image]

HomeAssistantPim, 2025-05-31:
@FireSon I was away, never the less as it seems the integration is not working for me with the solarflow 2400AC.
I only get sensor readings, can't change the limits (they just bounce back).
Also I don't see the P1 meter reading with the Zendure Manager, I do see the values coming in in the logging (see attached).
I tried to reconfigure the integration, but this fails without warnings either (just clears the form).
I've reached out to Zendure about linking a P1 meter via their app or at least to enable HEMS such that I can also create the second account.
Hence, still very much stuck and no solution yet.
Once I have more info from zendure, I probably going to setup the integration from scratch and maybe reach out in a new issue.

[home-assistant_zendure_ha_2025-05-31T18-08-09.394Z.log](https://github.com/user-attachments/files/20535181/home-assistant_zendure_ha_2025-05-31T18-08-09.394Z.log)