Zendure/Zendure-HA #297: Aggregation sensors are zero

MacdiverOne, 2025-05-24:
All aggregation sensors remain at zero since 41

Wunni73, 2025-05-24:
[image]

Yes it don't work in V41

surfer1264, 2025-05-24:
I recommend to build theses sensors on your own responsibility. It’s very easy to build energy sensors with the Helfer classes . (Integralsensor) 

At the end you are not depending from the integration. The only thing you need are the power sensors delivered by Z-HA.

https://github.com/surfer1264/Zendure-Stuff/wiki/yaml-Profi-Tricks-f%C3%BCr-Zendure#anlegen-der-integralsensoren

harrymayr, 2025-05-24:
> All aggregation sensors remain at zero since 41

Already a PR included, but not yet an Release.