Zendure/Zendure-HA #301: BypassMode mit AIO2400 steuerbar?

mloedl, 2025-05-25:
Hallo,

mein AIO2400 ist bislang nur passiv in HA, damit ich seine Werte dort weiterverwende. Verbindung über Zendure Cloud.
Seit dem letzten Update sollte auch der BypassMode steuerbar sein, bei mir wird er aber nur als Sensor ausgegeben, ohne Änderungsmöglichkeit.
Läuft das nur über lokalem MQTT?
[image]

Nebenbei: bei Geräteinfo wird jetzt eine falsche Bezeichnung angezeigt...?!
[image]

Danke für eventuelle Tipps.

harrymayr, 2025-05-25:
Problem ist, dass wahrscheinlich noch keiner das testen konnte und es 2 Variablen gibt, die das für die verschiedenen Devices steuern. Beim Hyper heißt das z.B. `gridReverse`, bei der Hub-Familie `passMode`.

Wenn du die Möglichkeit hast, die Datei \config\custom_components\zendure_ha\devices\aio2400.py zu ändern, könntest du das so wie unten angegeben ändern und dann testen, welche der beiden Variablen dann wirklich den Bypass ändert (und dann Bescheid geben ;-)  ).
Dann kann das eingebaut werden.
Bitte auf die Einrückung achten.

        selects = [
            self.select("acMode", {1: "input", 2: "output"}, self.update_ac_mode),
            self.select("passMode", {0: "auto", 2: "on", 1: "off"}),
            self.select("gridReverse", {0: "auto", 1: "on", 2: "off"}),
        ]

[image]

harrymayr, 2025-05-25:
Wahrscheinlich ist es aber passMode, da die Variable ja bei dir auftaucht.

mloedl, 2025-05-25:
Danke für die schnelle Reaktion!
Es ist tatsächlich passMode. (gridReverse bewirkt keine Änderung am Speicher).

Allerdings ändert sich auch der Wert von "autoRecover" zu 1, also zu "Am nächsten Tag auf Automatik umstellen", sofern er vorher in der App deaktiviert war. Kommt vermutlich von Zendure.
[image]

Ich plane täglich vor Sonnenaufgang abzuschätzen, ob ein Bypass-Betrieb sinnvoll ist und möchte das ausgeben. Dann ist die Einstellung von autoRecover für mich nicht relevant. 
Aber vielleicht kann der Wert über HA anpassbar sein?

und wenn wir schon dabei sind: 
kann der Energieplan (in HA "Hyper Energy Programm" / sensor.zendure_aio_2400_auto_model) im Cloud-Modus auch gewechselt (insb. deaktiviert) werden? 
Hier über App deaktiviert und wieder auf CT:
[image]

Zur zweiten Frage, hier die Batterie: 
[image]
(In der Zendure App wird kein Batterietyp angezeigt, ist AB1000 korrekt?)

harrymayr, 2025-05-25:
> Aber vielleicht kann der Wert über HA anpassbar sein?

Mit der Änderung müsste das funktionieren:

        selects = [
            self.select("acMode", {1: "input", 2: "output"}, self.update_ac_mode),
            self.select("passMode", {0: "auto", 2: "on", 1: "off"}),
            self.select("autoRecover", {0: "off", 1: "on"}),
        ]

> kann der Energieplan (in HA "Hyper Energy Programm" / sensor.zendure_aio_2400_auto_model) im Cloud-Modus auch gewechselt (insb. deaktiviert) werden?

Der Zendure Manager nutzt die Energiepläne (besser gesagt den CT-Mode) für den Smart-Mode.
Ob von @FireSon geplant ist, die App komplett nachzubauen (und damit auch das Schalten der Energiepläne) kann ich nicht sagen. Aber der Zweck der Integration ist vor allem, die Leistung über mehrere Geräte zu verteilen und mehr Kontrolle über die Devices zu haben.

harrymayr, 2025-05-27:
Die neue Version 1.0.44 sollte jetzt die Änderungen beinhalten.

closed 2025-05-28 by FireSon