Zendure/Zendure-HA #307: Smarte Leistungsregelung - p1meter

Stoppen14, 2025-05-25:
Hallo,

Ich habe mit der Konfiguration des Smart Modus Probleme.
Ich kann bei der Auswahl des p1meter keinen anderen sensor auswählen.

Ich habe eine Leistungssensor von Optipower.
Siehe Bilder. Er liefert wie in der Dokumentation angegeben negative Werte in W bei Einspeisung.

Wenn ich die ID in das Feld kopiere wird es nicht übernommen.

Woran könnte es liegen?

Danke. 

[image]
[image]

harrymayr, 2025-05-25:
Nicht über Konfigurieren, sondern über die 3 Punkte und Neu konfigurieren gehen

[image]

harrymayr, 2025-05-25:
Nach der Änderung aber auch HA neu starten.

Stoppen14, 2025-05-25:
> Nach der Änderung aber auch HA neu starten.

Habe ich getan. Leider klappt es noch nicht so recht. Muss man noch mehr einstellen?

[image]

harrymayr, 2025-05-25:
Wenn du ein Debug-Log erzeugt, kann ich mal schauen. Wenn du bei der Konfiguration localMQTT ausgewählt hast, würde ich erst mal ohne versuchen.

Stoppen14, 2025-05-25:
> Wenn du ein Debug-Log erzeugt, kann ich mal schauen. Wenn du bei der Konfiguration localMQTT ausgewählt hast, würde ich erst mal ohne versuchen.

Danke,

Ich glaube es klappt jetzt. Ich gebe morgen noch mal eine Rückmeldung.

Nochmals danke für die Hilfe.

[image]

Stoppen14, 2025-05-27:
@harrymayr es klappt bisher gut, danke für die Hilfe.

harrymayr, 2025-05-27:
Gerne. 😀
Gibt noch eine neue Version 1.0.44 mit ein paar kleinen Änderungen.
Für den Hyper im Wesentlichen, das die Verbleibene Restzeit sich jetzt am Min- und MaxSOC orientiert.

closed 2025-05-28 by FireSon