Zendure/Zendure-HA #316: 1.0.43 local MQTT connection

kutzera, 2025-05-26:
Local Mqtt Connection still Not working.

harrymayr, 2025-05-27:
On my end, it takes several minutes until the connection was established.
Please check out 1.0.44

kutzera, 2025-05-27:
> thx for this great hacs integration!
> 
> how do i found out if the hyper 2000 is visible via bluetooth to home assistant. i have bluetooth enabled and in the advertise monitor i can only see mac adresses. so i don´t know if the hyper 2000 is reachable to my home assistant.
> 
> can you please guide me into the wright direction, thx

Would be interested aswell because i just can’t get the connection to work

FireSon, 2025-05-27:
You should checkout the [wiki locak mqtt](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt) all steps are descibed.
@Bodo1981 If the device is recognized it shows BLE in the coonection status

FireSon, 2025-05-27:
@kutzera That is right, it also means that the connection between the device and your local MQTT is not working properly. 
What you should do is check if the connection between the device and the MQTT broker is succesful. 
If you use Mosquitto, you could delete the 'device users', they have exactly the same name as your deviceId. After that restart the integration. And check at the Mosquitto logging if there are connection errors. Also make sure you have entered the correct WiFi credentials...