Zendure/Zendure-HA #329: Use with evcc

peterfieber, 2025-05-28:
Thanks for your amazing Integration.

Since the MQTT / BLE Version is running, my Zendure app isnt updated regularly, like u mentioned in the wiki.
The Problem is, that evcc (EV charging AddOn for HA) uses the app to have the live PV Data. 

I guess the best solution now is to send evcc the live information via mqtt: HA->evcc AddOn every 1-20 seconds. 
Is there another option to have the live information in the app again?

Thank you for your nice work... this integration was one of the big pro reasons to buy Zendure ;)

harrymayr, 2025-05-28:
> I guess the best solution now is to send evcc the live information via mqtt: HA->evcc AddOn every 1-20 seconds.

I do it that way, because I send also other data (temperature of my water) to evcc.

FireSon, 2025-05-28:
@peterfieber Thanks nice to hear 😄 
Out of curiosity, does evcc connect to the Zendure MQTT server to collect the data, or does it use the Zendure API?

peterfieber, 2025-05-28:
evcc doesnt connect to the MQTT locally, they use the Cloud, like the app. 

I guess if somebody is into it it would be possible to implement the mqtt zendure <-> evcc. So its just a matter of time...

Raudi1, 2025-05-31:
Thanks for your template, @manuduarte . I had to add 
```
uri: https://homeassistant_.local...
insecure: true
```
since I don't have an SSL certificate for local access. Otherwise it seems to work perfectly.

Now I just have to figure out how to control the battery through EVCC. In the template there's the following example:
```
#batterymode: # set charging mode directly (1: normal, 2: hold, 3: charge) (for battery)
#  source: js
#  script: console.log(batterymode)
```
1 should be smart mode, 2 manual mode with 0W and 3 manual mode with -1200W

Raudi1, 2025-06-01:
I also noticed you only used the discharge value. I added charging as well like this:

```
power:
  source: calc
  add:
    - source: http
      uri: https://192.168.1.10:8123/api/states/sensor.hyper_2000_pack_input_power
      insecure: true
      headers:
      - content-type: application/json
      - Authorization: Bearer <token>
      jq: .state
      scale: 1
    - source: http
      uri: https://192.168.1.10:8123/api/states/sensor.hyper_2000_output_pack_power
      insecure: true
      headers:
      - content-type: application/json
      - Authorization: Bearer <token>
      jq: .state
      scale: -1
```

closed 2025-06-08 by FireSon