Zendure/Zendure-HA #335: local MQTT / help needed for AIO2400

Puncher4444, 2025-05-30:
Hello,

first of all, big thanks to @FireSon for this great piece of software - looks absolutly promising!
I was really happy once I´ve read that the switch to local MQTT should be possible now. So of course I directly tried it out but facing issues with integrating.
I´m using ESP32 BL proxy for BLE connection. In Bluetooth section in HA, the device is showing up correctly:

[image]

I also pressed 3 secons the IOT button on the AIO2400 and did I dont know how many restarts before and after setting up the integration, but nothing is working. In the MQTT log I dont see nothing. Also I dont see anything in the log from the integration itself - so I really dont know how to tackle this further. Any ideas?
Anyone else who got the AIO2400 working properly with local MQTT option?

PS: with the "native" cloud option all is working fine - but the aim is to get rid of the cloud connection and manage everything locally by myself.

Thanks
Daniel

FireSon, 2025-05-30:
Hi @Puncher4444 dirdyou check out the [local mqtt wiki](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt)? In there you can see that if the integration finds the bluetooth it shows up in the connection status:

[image]

So the first question is does it have this status? If not then, allthough it shows up in de BLE advertisment, it can be that the Integration cannot match it to an existing device! This is done by matching the serial number of the device with the manufacturer data in the advertisment. Can you check that?

Puncher4444, 2025-05-31:
Hi @FireSon 

the Connection Status shows "Unknown" (Unbekannt):

[image]

Of course I´ve read the wiki - and followed all the steps as well. But that dint really help in my case.
The advertisment for the Zen* device shows the following:

[image]

How should this by matching with the serial number of the device? As far as I see there isn´t any realtion between it.

From my point of view the issue is that the integration can not connect via BLE to the device. There is no indication in the logs that any BLE connection is being established. So if there should be any relation between the advetisment and the SN then this might be a reason.

Thanks
Daniel

FireSon, 2025-05-31:
@Puncher4444 @kutzera Thanks for the feedback, I think I found the problem, can you test it with version 45?

Puncher4444, 2025-05-31:
YES, its working now! Thanks a lot!

- 1st I took the AIO2400 into connection mode (pressing the IOT Button for 3 seconds)
- then I configured the integration
- after that was done, it was switching to connection state "BLE"
- then I restarted the AIO2400 (pressing the Power Button for 3 seconds - waited a bit and powered it back on -> pressing Power again for 3 seconds)
- then I switch "MQTTreset" and waited for a couple of secons after the first results came in

Thanks a lot! Perfect work - I´m really happy its working now!

closed 2025-05-31 by Puncher4444