Zendure/Zendure-HA #338: Step by step setup for bluetooth and local mqtt

wagnerpizza, 2025-05-31:
Sorry I dont get the system running with Bluetooth and mqtt.

I see the hub 2000 in bluetooth advertisment monitor.

I have a mosquito running with no existing users

I start the setup for integration of zendure and everything stays empty.

Do I have to press something on the hub or is some presetup of mosquito needed?

kutzera, 2025-05-31:
Wich version of the integration are you running? The 1.0.45 is working great

FireSon, 2025-05-31:
Here are Local mqtt: [instructions](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt)

CWILDMOUNTAIN, 2025-05-31:
>  The 1.0.45 is working great

Do you have to get the hub in "connection mode" by pushing the IoT button for 3 seconds (once or twice?) 

I did not manage to get a connection so far .. :(

CWILDMOUNTAIN, 2025-05-31:
2025-05-31 15:42:50.160 INFO (paho-mqtt-client-82898:5becb084-d65b-4038-a5f6-3bea2e52c48c) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /A8yh63/Hub 1/properties/report => {'messageId': '123', 'product': 'solarFlow', 'timestamp': 1748698968, 'properties': {'outputPackPower': 0, 'packState': 0}}
2025-05-31 15:42:50.555 INFO (paho-mqtt-client-82898:5becb084-d65b-4038-a5f6-3bea2e52c48c) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /8bM93H/Ace 1/properties/report => {'messageId': 908395, 'product': 'ace', 'timestamp': 1748698968, 'properties': {'solarInputPower': 11, 'solarPower1': 11}}
2025-05-31 15:42:51.608 INFO (paho-mqtt-client-82898:5becb084-d65b-4038-a5f6-3bea2e52c48c) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /8bM93H/Ace 1/properties/report => {'messageId': 908396, 'product': 'ace', 'timestamp': 1748698969, 'properties': {'solarInputPower': 9, 'solarPower1': 9}}
2025-05-31 15:42:52.277 INFO (paho-mqtt-client-82898:5becb084-d65b-4038-a5f6-3bea2e52c48c) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /8bM93H/Ace 2/properties/report => {'messageId': 909882, 'product': 'ace', 'timestamp': 1748698971, 'properties': {'solarInputPower': 116, 'solarPower1': 116}}
2025-05-31 15:42:52.659 INFO (paho-mqtt-client-82898:5becb084-d65b-4038-a5f6-3bea2e52c48c) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /8bM93H/Ace 1/properties/report => {'messageId': 908397, 'product': 'ace', 'timestamp': 1748698970, 'properties': {'solarInputPower': 10, 'solarPower1': 10, 'outputPackPower': 6, 'packState': 1, 'acHardwareVersion': 1, 'bmsHardwareVersion': 1}}
2025-05-31 15:42:53.323 INFO (paho-mqtt-client-82898:5becb084-d65b-4038-a5f6-3bea2e52c48c) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /8bM93H/Ace 2/properties/report => {'messageId': 909883, 'product': 'ace', 'timestamp': 1748698972, 'properties': {'solarInputPower': 117, 'solarPower1': 117}, 'packData': [{'maxVol': 336, 'sn': 'CO4HMCJMD800625'}]}
2025-05-31 15:42:53.671 INFO (paho-mqtt-client-82898:5becb084-d65b-4038-a5f6-3bea2e52c48c) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /A8yh63/Hub 2/properties/report => {'messageId': '123', 'product': 'solarFlow', 'timestamp': 1748698970, 'properties': {'solarInputPower': 269, 'outputPackPower': 366, 'solarPower2': 143}}

in the log file it seems to publish to the paho MQTT, but in the MQTT-Broker it’s not showing up .. 
any ideas?

wagnerpizza, 2025-05-31:
I used 1.0.44 and 45. and Bluetooth and mqtt isnt running. When I have to set the Hub into Connection Mode?

Puncher4444, 2025-05-31:
See https://github.com/FireSon/Zendure-HA/issues/335 - I‘ve described how I did it

wagnerpizza, 2025-05-31:
> Have you deleted the zendure users the persons section and restarted the zendure integration?

There are no users and persons. This is also one point I am wundering about.

CWILDMOUNTAIN, 2025-05-31:
> Have you deleted the zendure users the persons section and restarted the zendure integration?

Not so far. Did you?

Zanooon97, 2025-05-31:
also with the SF800 I have ble error in connection status

closed 2025-06-03 by FireSon