Zendure/Zendure-HA #339: ble error

Zanooon97, 2025-05-31:
Hi, now I have delete the topic for infinite reboot also on eu server with your new version 
Noow i trying to connect the local way.
I have put my raspberry 5 1,5m from the two SF800 and i am getting connection status BLE after restarting the integration.
But trying to reset mqtt getting immediately BLE error.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Zanooon97, 2025-06-01:
after letting itrhe night, it is lokal mqtt and zendure app

look:

[image]
[image]

kutzera, 2025-06-01:
I have the same Problem with my ACE1500 devices. I think the devices need to be programmed correctly inside the integration. Since the 1.0.45 my hub1200 device works via BLE. Maybe the solarflow und ACE will be added in the next version

FireSon, 2025-06-02:
As explained in the Local mqtt: [instructions](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt) there can be a lot of BLE errors. Most of the time they can be solved by moving the BLE adapter closer to the devices (or the other way around). 

If the Integration tries to communicate to the device and encounters some sort of error it will show BLE_ERR. Both of the errors I have had myself and were due to the devices being too far away. There is not much more I can do if you receive these errors, perhaps googling might give you some luck...

Trying a BLE amplifier might also help 😄

Zanooon97, 2025-06-04:
i will try it in future

FireSon, 2025-06-04:
@Zanooon97 thanks for all the feedback! In the mean time we will continue to try and resolve these BLE errors. There are more users who report these errors, even when the BLE adapter is within 1 meter of the devce. If there are any findings in that area I will let you know.
In version 47 some improvements are done to retry the commands after 5 minutes, and the connection status reflects better the actual connections.

closed 2025-06-04 by FireSon