Zendure/Zendure-HA #343: Regler  Hub 2000

rony2615, 2025-06-02:
Der Regler für die Ausgangsleistung geht nicht,  es lassen sich keine Werte einstellen.

harrymayr, 2025-06-03:
> Das Problem; stelle ich in der Zendure App das Ausgangslimit auf z.B. 100 und möchte es später in Homeassitant den Wert im Ausgangslimit ändern geht es nicht.

Aus meiner Sicht ist es sinnvoll in der App alle Energiepläne zu deaktivieren und mit der Zendure Integration zu steuern. Ausgangslimit in der App setzen, ist ein Energieplan.
Auch funktioniert meiner Meinung nach das Setzen eines manuellen Limits über die Manuelle Leistungsregelung besser, als über die Ausgangs- und Eingangslimit Slider.

harrymayr, 2025-06-03:
Ich habe nur einen Hyper, aber in der App kann ich dafür sorgen, dass keiner der Energiepläne aktiv ist. Ich denke beim Hub ist das genauso.
Und wenn bei euch die Regelung nicht funktioniert, helfen wir gerne. Pauschale Aussagen helfen da dann allerdings nicht, sondern Info darüber, welche Version installiert ist, ob lokalMQTT aktiviert ist und am besten ein Debug Log mit aktiviertem MQTTlogging. Dazu das Debug logging aktivieren und HA neu starten. Nach einigen Minuten das debug logging wieder deaktivieren, aus der heruntergeladenen Datei persönliche Informationen entfernen und hier einstellen.

[image]

z-master42, 2025-06-04:
Was ist denn mit Ausgangslimit gemeint? Das, welches man über Einstellungen - Netzein- & -ausgangseinstellungen festlegt?

rony2615, 2025-06-04:
Nein das Thema nicht abgeschlossen, habe mich vertan.

Linus86529, 2025-06-04:
@FireSon @harrymayr 
It may not work with a Hub 2000, but it works with my Hub1200. Please check out the attached screen copy. I have selected 66 Watt at the input field and after some seconds the sensor 'number.hub_1200_output_limit' shows constantly the value 66 Watt.

[image]

Linus86529, 2025-06-04:
Ich habe mit dem obigem Wert für das Ausgangslimit danach probiert was ausgegeben wird wenn batSOC < minSOC. Dann liefert 'sensor.hub_1200_output_home_power' = 0 W.