Zendure/Zendure-HA #345: Feedback on version 1.0.45 with the SolarFlow AC2400

HomeAssistantPim, 2025-06-02:
@FireSon some how it now seems to work.
I think this might be due to either upgrading the integration to version 1.0.45 (and a restart) or me setting a fixed schedule via the Zendure app.

I do how ever have two Zendure Managers now.
An old one that no longer has sensors and I'm not allowed to removed it.
The new one manages to set the power limits and toggle charge / discharge when needed.
The new one does not show the readings of the P1 as illustrated in https://github.com/FireSon/Zendure-HA/issues/287.

Also the batteries are listed seperately from the SolarFlow AC device and the battery sensors part of the device have become unavailable.

Happy to see that controlling the limits works, I do see room for improvement on the Zendure Manager (see image).
It seems to jitter and as all energy from/to the grid is charged with taxes, a better strategy is needed.
I will try and see if I can manage to create a better strategy and share it.

[image]

closed 2025-06-02 by HomeAssistantPim

FireSon, 2025-06-02:
Nice that is good to hear 😄 Could you do me a favour and check if you have somewhere in your debuglogging information regarding a ble_mac or something like it? That should be the bluetooth mac address. It would help with the bluetooth connection and possible use of local MQTT. This information is probably visible after start up, if you let it run for 5 minutes with debug logging on.  It would be best if you send it to j.j.fireson at gmail.com Thanks!!

FireSon, 2025-06-02:
O that is not necessary! 
Just as it is now is fine, the SF2400 will hopefully report the property