Zendure/Zendure-HA #347: Regular disconnect in app and integration

DeckardCain2014, 2025-06-03:
In the last two weeks I observed multiple times the app and the integration is disconnected from the Zendure cloud.
The LEDs show charging/discharging/empty battery, so the system itself works. But only restarting helps to connect it back to the cloud.
It shows connected however in the App, but the values are not updated. Also, it is 10m apart from the router, so it should not be a WiFi issue.

Sadly I can't show you the sensor history, as they all got renamed in the last update from `hyper2000[...]` to `hyper_2000[...]`

Debug log shows no error and also connection state 'connected'.

Reloading the integration in this state returns in unknown sensors.

```
2025-06-03 11:33:54.356 DEBUG (MainThread) [custom_components.zendure_ha] Open API connection
2025-06-03 11:33:54.356 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
2025-06-03 11:33:56.211 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/eu => Mqtt: mqtteu.zen-iot.com
2025-06-03 11:33:56.211 INFO (MainThread) [custom_components.zendure_ha.api] Getting device list ...
2025-06-03 11:33:56.406 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding device: FGs9xU5b Hyper 2000
2025-06-03 11:33:56.406 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Data: {'id': 170728, 'deviceKey': 'FGs9xU5b', 'snNumber': 'EE1LYMGJN010012', 'name': 'Hyper 2000', 'productId': 68, 'productKey': 'gDa3tb', 'onlineFlag': '00601', 'productName': 'Hyper 2000', 'wifiStatus': True, 'blueState': False, 'fourGStatus': False, 'isShareFlag': '00601', 'input': False, 'output': False, 'electricity': 39, 'hasPack': True, 'restState': 0, 'upsMode': False, 'upgradeStatusDes': '', 'productType': 17, 'deviceType': 0, 'upgradeStatus': {}, 'bindId': 0, 'bindStatus': 0, 'batteryCode': '', 'packList': [], 'inputPower': 0, 'outputPower': 0, 'slowChargePower': 0, 'temperature': 0, 'temperatureUnit': 0, 'remainOutTime': 0, 'bindType': 0, 'seriesMode': 0, 'parallelMode': 0, 'networkType': 0, 'standard': 'E', 'isSwitch': False, 'url': '', 'remainOil': 0, 'genMode': 0, 'shortCode': 'E1', 'phaseCheck': 0, 'pass': 0, 'clusterList': [], 'clusterNotice': False, 'nodeStatus': 0, 'clusterId': None, 'phase': None, 'reverseState': 0, 'meterId': 0, 'hemsId': 0, 'totalPower': 0, 'keyFlag': 0}
2025-06-03 11:33:56.672 DEBUG (MainThread) [custom_components.zendure_ha.sensor] Restored state for sensor.hyper_2000_aggr_charge_total: 5.56002393960819
2025-06-03 11:33:56.672 DEBUG (MainThread) [custom_components.zendure_ha.sensor] Restored state for sensor.hyper_2000_aggr_discharge_total: 4.8631989006184
2025-06-03 11:33:56.672 DEBUG (MainThread) [custom_components.zendure_ha.sensor] Restored state for sensor.hyper_2000_aggr_solar_total: 13.8445714736884
2025-06-03 11:33:56.691 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Found: 1 devices
2025-06-03 11:33:56.692 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding sensors Zendure Manager
2025-06-03 11:33:56.692 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 0
2025-06-03 11:33:56.692 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Energy sensors: sensor.power_actual to _update_smart_energyp1
2025-06-03 11:33:56.709 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Zendure Manager initialized
2025-06-03 11:33:56.709 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-06-03 11:33:56.709 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching Zendure Manager data in 0.001 seconds (success: True)
2025-06-03 11:35:26.835 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
2025-06-03 11:35:26.835 DEBUG (MainThread) [custom_components.zendure_ha.zendurermanager] Finished fetching Zendure Manager data in 0.001 seconds (success: True)
```

[image]

[image]

DeckardCain2014, 2025-06-03:
Thanks for that addition! Can you share the complete automation code, I would really like to adopt it. How do you disconnect it remotely?

FireSon, 2025-06-04:
In version 47 I have added regular updates each 5 minutes from the Integration to the App. Today the App is still working properly, so the disconnect could also be caused by 'inactivity'. Please check if with version 47 it also works for you!

closed 2025-06-05 by FireSon

DeckardCain2014, 2025-06-06:
Today it started again, until now I didn't get it working again ;(