Zendure/Zendure-HA #348: Hyper 2000 : no data in Home Assistant despite everything seems to be working fine

nicolinuxfr, 2025-06-03:
Hello

I'm trying to use the local MQTT server with my Hyper 2000 and even if everything seems fine on the log side (see below), I have no data and nothing works on Home Assistant.

```
homeassistant  | 2025-06-03 18:39:08.016 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Found: 1 devices
homeassistant  | 2025-06-03 18:39:08.016 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding sensors Zendure Manager
homeassistant  | 2025-06-03 18:39:08.017 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new select.zendure_ha entity: select.zendure_manager_operation
homeassistant  | 2025-06-03 18:39:08.018 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update operation: 0 from: 0
homeassistant  | 2025-06-03 18:39:08.020 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new number.zendure_ha entity: number.zendure_manager_manual_power
homeassistant  | 2025-06-03 18:39:08.023 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Energy sensors: sensor.garage_compteur_shelly_em2_power to _update_smart_energyp1
homeassistant  | 2025-06-03 18:39:08.078 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Use local MQTT broker
homeassistant  | 2025-06-03 18:39:08.081 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Zendure Manager initialized
homeassistant  | 2025-06-03 18:39:08.082 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
homeassistant  | 2025-06-03 18:39:08.082 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Found Zendure Bluetooth device: <BluetoothServiceInfoBleak name=ZenE1_97 address=94:C9:60:D5:D7:89 rssi=-60 manufacturer_data={17733: b'1LYMGJM320797.'} service_data={} service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000a002-0000-1000-8000-00805f9b34fb', '00112233-4455-6677-8899-aabbccddeeff'] source=00:1A:7D:DA:71:13 connectable=True time=868517.660729657 tx_power=None raw=None>
homeassistant  | 2025-06-03 18:39:08.172 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager iot/gDa3tb/Hyper 2000/properties/read => {'properties': ['getAll']}
homeassistant  | 2025-06-03 18:39:08.172 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager iot/gDa3tb/Hyper 2000/properties/read => {'properties': ['getAll']}
homeassistant  | 2025-06-03 18:40:38.150 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
homeassistant  | 2025-06-03 18:40:38.151 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager iot/gDa3tb/Hyper 2000/properties/read => {'properties': ['getAll']}
homeassistant  | 2025-06-03 18:42:08.149 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
homeassistant  | 2025-06-03 18:42:08.150 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager iot/gDa3tb/Hyper 2000/properties/read => {'properties': ['getAll']}
homeassistant  | 2025-06-03 18:43:38.149 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] refresh devices
homeassistant  | 2025-06-03 18:43:38.150 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager iot/gDa3tb/Hyper 2000/properties/read => {'properties': ['getAll']}
```

The status on HA is "Unknown and BLE", which is not great. I'm guessing the Hyper 2000 is never connected to the Wi-Fi but I checked, the informations I gave the integration are correct and it should work. I have seen no error in the logs.

[image]

What should I do ?

harrymayr, 2025-06-03:
In the Log I can see, that the Hyper does not answer on the getAll request. Could you try (perhaps several times) MqttReset?

TheHangMan97, 2025-06-03:
Same here :/

[image]

FireSon, 2025-06-03:
Did you follow the Local mqtt: [instructions](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt)
Which MQTT broker do you use?
Did you add the MQTT integration?

nicolinuxfr, 2025-06-04:
> Did you follow the Local mqtt: [instructions](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt) Which MQTT broker do you use? Did you add the MQTT integration?

Of course ! 🙂 

I can see that it's working on the MQTT side :

[image]

harrymayr, 2025-06-05:
There is now answer of the Hyper, you see only the MQTT request of the Zendure Integration.
Is the IP 192.168.1.8 the same IP where you reach HA?

I'm asking, because my host for the MQTT broker is not an IP

[image]

nicolinuxfr, 2025-06-05:
Yes, it's the correct IP, no problem there. 

I think the issue is with the Wi-Fi connection, but I'm not sure why. When I connect the Hyper 2000 to the same network, it works fine but using the integration, the device seems stuck on Bluetooth.

harrymayr, 2025-06-05:
> When I connect the Hyper 2000 to the same network, it works fine but using the integration, the device seems stuck on Bluetooth.

What does this mean?? Is the Hyper not always connected to the Network?
The integration only uses BT to change the communication from the Zendure Server to the own local MQTT-Server. But it needs WiFi for both connections.

nicolinuxfr, 2025-06-05:
> What does this mean?? Is the Hyper not always connected to the Network?

Sorry, I forgot some words : it works fine using the Zendure app. 

And I was wrong, it seems the Hyper 2000 is indeed connected to the Wi-Fi even using this integration, so nevermind. I don't understand why it's not working then… 🤔

The only data I have inside MQTT is the getAll :

[image]

nicolinuxfr, 2025-06-05:
I just thought of something and checked the MQTT logs. I think it's an authentification issue : 

```
mqtt  | 1749109156: New connection from 192.168.1.100:64360 on port 1883.
mqtt  | 1749109157: Client d8bbkR81 disconnected, not authorised.
```

I'm not sure why, shouldn't this integration use the credentials from Home Assistant's MQTT integration ? Anyway, my MQTT server is indeed password protected, so it would makes sense if it does not receive the correct informations.

harrymayr, 2025-06-05:
But that means, your HA instance is running on 192.168.1.100, your MQTT Server on 192.168.1.8

@FireSon: can this work?

FireSon, 2025-06-05:
@nicolinuxfr The "Getall" topic is written by the Integration. 
I suspect that there might be an authentication issue. Which broker do you use and is Authentication in place for it? If it is Mosquitto, it might help to delete the 'device users' from HA. You can also check the Mosquitto logging page.

@harrymayr that is no problem, as long as the authentiaction is working!

FireSon, 2025-06-05:
@nicolinuxfr If you have Mosquitto, and I thought I recognized the logging, it is best to delete the existing device users, they will be automatically added again with the correct password. Otherwise, you ca add the users manually to the broker, for the Zendure devices the username is de deviceId and the passwords you can find in the debuglogging of the Integration, when it is (re)started.

nicolinuxfr, 2025-06-05:
> But that means, your HA instance is running on 192.168.1.100, your MQTT Server on 192.168.1.8

No, the Hyper 2000 is the 192.168.1.100 device in this scenario. HA is running on the same IP as MQTT.

> I suspect that there might be an authentication issue. Which broker do you use and is Authentication in place for it? If it is Mosquitto, it might help to delete the 'device users' from HA. You can also check the Mosquitto logging page.

Sorry, I should have answer it earlier. I'm using Mosquitto indeed, with password authentification. And I think the issue may be that I connected the Zendure cloud MQTT in my config, I'm guessing there is a conflict with this integration. 

```
connection zendure-broker
address mqtt-eu.zen-iot.com:1883
remote_username XXXX
remote_password XXXX
remote_clientid 1d4n8C57
topic 1d4n8C57/# in
#topic # in 0 homeassistant/sensor/1d4n8C57/ 1d4n8C57/sensor/device/
#topic # in 0 homeassistant/switch/1d4n8C57/ 1d4n8C57/switch/device/
```

Should I remove this lines and reconfigure ? Or do something else ?

Thanks !

FireSon, 2025-06-05:
Yes that should be your local MQTT connection

nicolinuxfr, 2025-06-05:
I removed everything, reconfigured, but same. In the log, I noticed the weird port number which keeps increasing, is it supposed to do that ?

```
mqtt  | 1749128658: New connection from 192.168.1.100:59403 on port 1883.
mqtt  | 1749128658: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749128658: New connection from 192.168.1.100:59404 on port 1883.
mqtt  | 1749128658: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749128658: New connection from 192.168.1.100:59405 on port 1883.
mqtt  | 1749128658: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749128658: New connection from 192.168.1.100:59406 on port 1883.
mqtt  | 1749128658: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749128658: New connection from 192.168.1.100:59407 on port 1883.
mqtt  | 1749128658: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749128659: New connection from 192.168.1.100:59408 on port 1883.
mqtt  | 1749128659: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749128659: New connection from 192.168.1.100:59409 on port 1883.
mqtt  | 1749128659: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749128659: New connection from 192.168.1.100:59410 on port 1883.
mqtt  | 1749128659: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749128659: New connection from 192.168.1.100:59411 on port 1883.
mqtt  | 1749128659: Client d8bbkR81 disconnected, not authorised.
```

TheHangMan97, 2025-06-05:
[image]

Is see this in the log, but not at the devices

[image]

i have 2 Hyper

harrymayr, 2025-06-05:
Give the Hyper different names in the App. (In the settings, small green pen at the top)
The restart HA.

TheHangMan97, 2025-06-05:
> Give the Hyper different names in the App. (In the settings, small green pen at the top) The restart HA.

**

[image]

**

harrymayr, 2025-06-05:
And now check in Einstellungen / Geräte & Dienste / Zendure Home Assistant..
My experiance ist, that the devices are showed with the names you marked in the Log.

TheHangMan97, 2025-06-05:
> And now check in Einstellungen / Geräte & Dienste / Zendure Home Assistant.. My experiance ist, that the devices are showed with the names you marked in the Log.

[image]

TheHangMan97, 2025-06-05:
> Still no values on the devices?

only this

[image]

TheHangMan97, 2025-06-05:
> > Still no values on the devices?
> 
> only this
> 
> [image]

[image]

harrymayr, 2025-06-05:
You seem not to have a bluetooth connection to your devices. Please disable localMQTT in the configuration (3 dots and then "Neu Konfigurieren") and restart HA.

TheHangMan97, 2025-06-05:
> You seem not to have a bluetooth connection to your devices. Please disable localMQTT in the configuration (3 dots and then "Neu Konfigurieren") and restart HA.

But i can see it

[image]

TheHangMan97, 2025-06-05:
> Is the scanner-mode of the ble repeater active or passive? I assume it must be active to work with the integration.

[image]

nicolinuxfr, 2025-06-05:
On my side, I started over, deleting everything and re-adding my Hyper 2000. As far as I can tell from the logs, everything is fine :

```
homeassistant  | 2025-06-05 22:07:39.736 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
homeassistant  | 2025-06-05 22:07:40.614 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/eu => Mqtt: mqtteu.zen-iot.com
homeassistant  | 2025-06-05 22:07:40.615 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up zendure_ha.binary_sensor
homeassistant  | 2025-06-05 22:07:40.615 INFO (MainThread) [homeassistant.components.number] Setting up zendure_ha.number
homeassistant  | 2025-06-05 22:07:40.616 INFO (MainThread) [homeassistant.components.select] Setting up zendure_ha.select
homeassistant  | 2025-06-05 22:07:40.616 INFO (MainThread) [homeassistant.components.sensor] Setting up zendure_ha.sensor
homeassistant  | 2025-06-05 22:07:40.616 INFO (MainThread) [homeassistant.components.switch] Setting up zendure_ha.switch
homeassistant  | 2025-06-05 22:07:40.616 INFO (MainThread) [custom_components.zendure_ha.api] Connecting to Zendure
homeassistant  | 2025-06-05 22:07:41.437 INFO (MainThread) [custom_components.zendure_ha.api] Connected to https://app.zendure.tech/eu => Mqtt: mqtteu.zen-iot.com
homeassistant  | 2025-06-05 22:07:41.437 INFO (MainThread) [custom_components.zendure_ha.api] Getting device list ...
homeassistant  | 2025-06-05 22:07:41.598 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Adding device: d8bbkR81 Hyper 2000
homeassistant  | 2025-06-05 22:07:41.598 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Data: {'id': 128497, 'deviceKey': 'd8bbkR81', 'snNumber': 'EE1LYMGJM320797', 'name': 'Hyper 2000', 'productId': 68, 'productKey': 'gDa3tb', 'onlineFlag': '00602', 'productName': 'Hyper 2000', 'wifiStatus': False, 'blueState': False, 'fourGStatus': False, 'isShareFlag': '00601', 'input': False, 'output': False, 'electricity': 54, 'hasPack': True, 'restState': 0, 'upsMode': False, 'upgradeStatusDes': '', 'productType': 17, 'deviceType': 3, 'upgradeStatus': {}, 'bindId': 0, 'bindStatus': 0, 'batteryCode': '', 'packList': [], 'inputPower': 0, 'outputPower': 0, 'slowChargePower': 0, 'temperature': 0, 'temperatureUnit': 0, 'remainOutTime': 0, 'bindType': 0, 'seriesMode': 0, 'parallelMode': 0, 'networkType': 0, 'standard': 'E', 'isSwitch': False, 'url': '', 'remainOil': 0, 'genMode': 0, 'shortCode': 'E1', 'phaseCheck': 0, 'pass': 0, 'clusterList': [], 'clusterNotice': False, 'nodeStatus': 0, 'clusterId': None, 'phase': None, 'reverseState': 0, 'meterId': 0, 'hemsId': 0, 'totalPower': 0, 'keyFlag': 0}
homeassistant  | 2025-06-05 22:07:41.886 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Created MQTT user: d8bbkR81 with password: EFBAAAC81ADFC56E
homeassistant  | 2025-06-05 22:07:41.887 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zendure_ha entity: sensor.hyper_2000_aggr_charge_total
homeassistant  | 2025-06-05 22:07:41.893 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zendure_ha entity: sensor.hyper_2000_aggr_discharge_total
homeassistant  | 2025-06-05 22:07:41.899 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zendure_ha entity: sensor.hyper_2000_aggr_solar_total
homeassistant  | 2025-06-05 22:07:41.903 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.zendure_ha entity: sensor.hyper_2000_connection_status
homeassistant  | 2025-06-05 22:07:41.908 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new switch.zendure_ha entity: switch.hyper_2000_mqtt_reset
homeassistant  | 2025-06-05 22:07:41.913 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_master_switch
homeassistant  | 2025-06-05 22:07:41.920 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_buzzer_switch
homeassistant  | 2025-06-05 22:07:41.924 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_wifi_state
homeassistant  | 2025-06-05 22:07:41.930 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_heat_state
homeassistant  | 2025-06-05 22:07:41.932 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_reverse_state
homeassistant  | 2025-06-05 22:07:41.935 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_pass
homeassistant  | 2025-06-05 22:07:41.937 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_low_temperature
homeassistant  | 2025-06-05 22:07:41.939 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_auto_heat
homeassistant  | 2025-06-05 22:07:41.942 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new binary_sensor.zendure_ha entity: binary_sensor.hyper_2000_local_state
[... cut ...]
But the Hyper 2000 device in Home Assistant is still full of unknown entities and I can't control it. And in the mosquitto logs, I still see many messages like this one : 

```
mqtt  | 1749154770: New connection from 192.168.1.100:52562 on port 1883.
mqtt  | 1749154770: Client d8bbkR81 disconnected, not authorised.
```

I'm not sure what I should do next. Is this a mosquitto config error ? Here it is, if it can help : 

```
allow_anonymous false
listener 1883
listener 9001
protocol websockets
persistence true
password_file /mosquitto/config/pwfile
persistence_location /mosquitto/data/
```

harrymayr, 2025-06-05:
> I don't see this line :( i'm using a ble repeater... maybe this isn't working with the addon? My Plant Sensors are working fine with the ble repeater... and the repeater see this Zen Devices..

@TheHangMan97 OK, I can confirm, that it also does not show BLE if I use my Shelly 4PM as a BLE gateway and disable the BLE of the Raspberry.
The BLE device is seen, but it shows `connectable=False` and it show also less `service_uuids` than the Raspi BLE. Have not yet further ideas. Maybe you see the lines with `[custom_components.zendure_ha.zendurermanager] Found Zendure Bluetooth device: <BluetoothServiceInfoBleak ...` if you select MQTT Reset, but not sure.

nicolinuxfr, 2025-06-06:
> Perhaps the protocol is the problem.

Good idea, I tried, but no change.

While seeing all the logs, here's what I have. Not much better, I'm afraid.

```
mqtt  | 1749189056: New connection from 192.168.1.100:55456 on port 1883.
mqtt  | 1749189056: Sending CONNACK to d8bbkR81 (0, 5)
mqtt  | 1749189056: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749189056: New connection from 192.168.1.100:55457 on port 1883.
mqtt  | 1749189056: Sending CONNACK to d8bbkR81 (0, 5)
mqtt  | 1749189056: Client d8bbkR81 disconnected, not authorised.
mqtt  | 1749189056: New connection from 192.168.1.100:55458 on port 1883.
mqtt  | 1749189056: Sending CONNACK to d8bbkR81 (0, 5)
mqtt  | 1749189056: Client d8bbkR81 disconnected, not authorised.
```

nicolinuxfr, 2025-06-15:
FYI, I managed to make it work, by adding the user and password for Zendure integration directly into my MQTT password file. I'm closing the issue but I can help other if need be, let me know.

closed 2025-06-15 by nicolinuxfr