Zendure/Zendure-HA #359: Impossible to force AC Input mode on SolarFlow 2400 AC — immediately switches back to AC Output

wpformation, 2025-06-07:
Hi,
First of all, thank you for your amazing work on this integration!

I’m using a Zendure SolarFlow 2400 AC with the latest version of your integration via HACS on Home Assistant (HAOS). The integration is working well in terms of telemetry, and I'm using the main Zendure account (not a shared one).

Problem
When I try to force the AC Input mode (either manually via the select.ac_mode entity or using a service call), the device briefly accepts the change, and then switches back instantly to AC Output, without error.

Here’s what I’ve already tested:

zendure_manager is active and set to Manual
number.zendure_manager_manual_power is set to +1000
soc_max lowered to 75%
Tried cutting AC Output physically (via circuit breaker)
Set app energy plan to "Basic charging mode"
Fully logged out of the Zendure mobile app
Rebooted HA and reloaded the integration
Each time, when I switch the ac_mode to "AC Input", the device:
Accepts the change (writeRsp = 2)
Switches back to "AC Output" within 2–3 seconds

Notes
This issue seems specific to the SolarFlow 2400 AC
Other users report the same behavior on forums and Discord (HACF & Zendure)
It may be a firmware-level restriction that overrides external AC Input requests, even in Manual mode

Request
Could you confirm:

Is this behavior expected on the 2400 AC?
Does your integration currently handle that model differently from Hyper 2000 or SolarFlow 800?
Is there any workaround you recommend to truly force AC Input mode?

Thanks again for your great work — happy to help test or provide logs if needed!

harrymayr, 2025-06-07:
For AC Input you can use a negative value in manual mode.

wpformation, 2025-06-07:
> For AC Input you can use a negative value in manual mode.

I've tried but no success :/