Zendure/Zendure-HA #365: Hub 2000 /ACE 1500 Keine Lokales Mqtt oder BLE

PennyOttah, 2025-06-08:
Hallo zusammen 

ich habe folgendes Probleme  ich bekomme die lokalen mqtt oder die BLE Verbindung nicht zum laufen Cloud läuft ohne Problem, es fängt schon das ich meine Geräte nicht Im HA  im BL Advertise-Monitor sehe  ich habe Direkt  neben der Anlage ein sehlly im Bluetooth-Scannermodus auf Aktiv ca 1m 

danke Im voraus und großes lob an die Entwickler die Integration ist super  :D

Linus86529, 2025-06-08:
Das ist aber richtig blöd.

TheHangMan97, 2025-06-08:
Same Problem with Shelly BLE Repeater

PennyOttah, 2025-06-08:
und was kann Mann machen wegen der BLE Verbindungen  ist es den mit shelly möglich

harrymayr, 2025-06-09:
> und was kann Mann machen wegen der BLE Verbindungen ist es den mit shelly möglich

Ich habe jetzt verschiedene Versuche mit einem Shelly 4PM gemacht. Mir ist es nicht gelungen darüber eine Verbindung aufzubauen.
Immer wenn ich von meinem Raspi das BT-Device in HA deaktiviere sehe ich zwar den Hyper im Advertise-Monitor beim Shelly, wenn ich dann aber die Zeile anklicke und mir die Geräte-Daten in die Zwischenablage kopiere und in einem Editor anschaue, finde ich immer den Eintrag "connectable":false drin. Ich habe keine Idee, wie man einen Shelly im BLE Gateway-Modus dazu bewegen kann "connectable":true zu zeigen.