Zendure/Zendure-HA #373: Is it possible/wise to change the reaction time in smart matching mode

HomeAssistantPim, 2025-06-10:
Smart matching works nice but in my setup it is lagging 6 sec to adjust the power limit (I have not tried local MQTT yet).
With heating devices and solar panels it would be cool if the response would be a bit quicker (6s = 1 kWh / day here).
Is this delay caused by my Solarflow or is this an interval set on the Zendure Manager?
In case of the latter, would it be possible/wise to increase the interval to match the updates of my P1 reading (which I force to be at 1 sec)?

harrymayr, 2025-06-11:
Depends which device you have. The HUB family is pretty slow in increasing some load out of the battery, this could not changed by SW. For a hyper or other devices you can play with the times defined in const.py  (TIMEFAST, TIMEZERO)

harrymayr, 2025-06-11:
With a Hyper and both times set to 2, I have less than 5Wh in 7 hours

[image]

FireSon, 2025-06-11:
@HomeAssistantPim The integration reacts within 3 seconds to a change > 100w and within 5 seconds to a smaller one. But it also depends on your P1 meter update frequency. It is possible to update sooner but as @harrymayr said, there are devices which are slower, so it might be something which we could make device specific if needed!