Zendure/Zendure-HA #376: 2 Hyper 2K - both broken-down - can't delete one of them

tootai, 2025-06-11:
Hi I have 2 hypers in cluster, both died. I could remove one not the other one. Message is:

> Failed to remove device entry, rejected by integration

What does it mean?

FireSon, 2025-06-11:
You cannot delete indivdual devices from the Integration. All the device from you account are added automatically. If you really want to remove a device you have to do if from the App. 
But maybe it is better to remove the Hyper2000 from the grid and from the batteries, wait 10 minutes and out them back together. Normally it should work again...
I hope you can fix them!

tootai, 2025-06-11:
Both hypers no more appears in app, only Shellys. Should be something else ...

tootai, 2025-06-12:
[image]
[image]
[image]

Screenshots you ask for. Can't say or do more :(

FireSon, 2025-06-12:
Ok the last one is interesting! Do you use a second account for the Integration? Most users do, and then you have to 'share' your device to this account. Could it be that your device sharing is not correct anymore? That could explain the problem!

tootai, 2025-06-12:
You're right, all shared addresses disappears from sharing! A way to solve this?

FireSon, 2025-06-12:
Yes, you have to share them again 😄