Zendure/Zendure-HA #378: Memory flash

Spottttt, 2025-06-12:
Bonjour @FireSon 

Que pouvez me dire sur l'utilisation des mode Acinput/output et la saturation de la mémoire flash ??
Est que cela veut dire que, à terme, cette saturation provoquera des dysfonctionnements et qu'il vaut mieux utiliser le smart mode de l'intégration??

#################################################

Hello @FireSon 

What can you tell me about using AC input/output modes and flash memory saturation?
Does this mean that, over time, this saturation will cause malfunctions and that it's better to use the integration's smart mode?

FireSon, 2025-06-12:
@Spottttt Nice that you come up with this question, since we more or less had a similar discussion somewhere else 😄 

When I started the Integration, I did nt want to use the AC mode, because of this reason. I thought if Zendure uses something different for frequent updates, I should do the same. That is the reason you see the device status changes to "Smart mode". And the integration uses the same smartmode commands as the CT smartmode of the device itself.

In order to fix a communication issue for the Hyper2000, I asked Zendure if that was OK to use it in that way, and this was their reply:

> I learned from the colleagues in device development that controlling through the AC mode is different from the autoModelProgram. The AC mode writes the power to the flash, and if commands are issued every 2 seconds, the flash will quickly become full, causing device malfunctions.

So that seemed a dead end. Then in a discussion about the new SF devices, the issue came up again and then I asked it once more and they responed:
> For the device, each time the output mode is switched, it will be written to the flash. Why is it written? It is designed to ensure that the device can maintain the last state after power failure. For the new 2025 products, you can find a property called smartMode. Once the value is 1, the ac mode setting will no longer be written to the flash, so the flash life will be much longer.
> Therefore, for linkage in ct mode, smartmode will be 1, and acmode can be set repeatedly to achieve a balance in overall household energy input and output.

So since the integration is using the same commands as the smart mode of the device, there should not be a problem... At least I am confident to use it myself 😉

Spottttt, 2025-06-12:
@FireSon 
Anticipant votre réponse, j'ai déjà commencé la migration de mes automatisations vers le Smart mode du Zendure manager.
Et ma foi, cela marche plutôt bien aussi .

Merci pour la clarification

###############################################################

@FireSon 
Anticipating your response, I've already started migrating my automations to Zendure Manager's Smart mode.
And, by golly, it's working pretty well too.

Thanks for the clarification.