Zendure/Zendure-HA #386: Hyper 2000 BLE Problem

Z3R0RAG3, 2025-06-14:
Wie einige hier ja schon berichten, kommt es auch bei mir jeden Tag zu Problemen mit dem Hyper 2000 in Verbindung mit der Integration.

Der Hyper springt leider immer wieder auf die BLE-Only-Verbindung. Nach einigem Hin und Her habe ich dann nach Deaktivierung der Integration herausgefunden, dass es an dieser liegt.

Deaktiviere ich die Integration, läuft der Hyper völlig problemlos!
Ist diese dagegen aktiviert, verliert er spätestens um 0:00 Uhr nachts die Verbindung zur Cloud und zum WLAN. In der Folge muss er dann komplett neu ins WLAN eingebunden werden.

Aktuell ist die einzige Lösung (trotz der Fix-Versuche) das Deaktivieren der Integration.

PS: Mein Hub 2000 läuft dagegen auch mit der Integration völlig problemlos.

In diesem Sinne, schöne Grüße

harrymayr, 2025-06-14:
Macht es einen Unterschied, ob du localMqtt auswählt, oder nicht?

Linus86529, 2025-06-15:
Moin @harrymayr,
auch bei mir sind heute wieder BLE-Fehler beim Hyper2k aufgetreten. Im Log wurde der folgende Block im Abstand von fünf Minuten wiederholt eingetragen. Ich nehme an, dass dies der Zeitraum ist, in dem versucht wird, den Fehlerzustand zu beseitigen.:
[image]

Die Abfolge der Verbindungs-Codes des 'Hyper 2000 Connection Status' ist hier zu sehen:
[image]

Die Verbindung zu meinen BT-Geräten ist stabil und die Signalstärke ist gut:
[image]

harrymayr, 2025-06-15:
Bist du sicher mit der P1_58? Ich würde glauben, P... ist ein ACE1500. Und alle anderen Logs die ich von dir habe sind E1_35 und E1_79.
Hast du mal abgeglichen, ob die Geister Akkus zu seinen Seriennummen passen???

Robia1, 2025-06-15:
Tritt bei mir auch auf. Ist mir erst aufgefallen nachdem ich 2 mal den Zendure Server gewechselt habe und dabei die Anbindung an das 2. Konto rausgeflogen ist. Interessant ist, wenn der Zendure Server auf global geändert wird, fliegt die WLAN Verbindung im Minutentakt raus und auf dem deutschen Server immer über Nacht. Der geht auf Bluetooth only. WLAN neu verbinden funktioniert max. 1Tag (deutscher Server). Zurücksetzen Hyper 10 Sekunden bringt auch nichts daher erstmal die Integration entfernt. Der Zendure Kundendienst schob erstmal auf WLAN Probleme bei mir aber hat auch was von einer API Problematik gesprochen, die allerdings nun behoben sein sollte.. Das war die Antwort als ich erwähnte das die Zendure Anbindung beim Serverwechsel herausgeflogen ist.

Barrage83, 2025-07-13:
ich hatte ebenso permanentes wechseln... beim Shelly muss das Bluetooth deaktiviert werden da deren ble gateway Standard aktiv ist... meine Vermutung:  da sich beide Geräte kennen aufgrund local WiFi Kommunikation versuchen sie sich zu verbinden... da im Shelly aber kein ble Gerät eingetragen ist wird Verbindungsversuch beendet und hyper geht wieder zur Cloud über .. seit der Deaktivierung von Bluetooth beim Shelly keinerlei Probleme mehr Status: Zendurecloud und BLE  durchgehend aktiv

Z3R0RAG3, 2025-07-16:
Leider hat das deaktivieren des Bluetooth des Shellys nur 2 Tage lang etwas gebracht.. 
Jetzt ist das selbe Problem wieder aktiv..

"16. Juli 2025
Hyper 2000 Connection Status geändert zu BLE
09:17:25 - Vor 2 Stunden
Hyper 2000 Connection Status geändert zu BLE Fehler
09:11:20 - Vor 2 Stunden
Hyper 2000 Connection Status geändert zu BLE
08:28:42 - Vor 3 Stunden
Hyper 2000 Connection Status geändert zu BLE Fehler
08:22:36 - Vor 3 Stunden
Hyper 2000 Connection Status geändert zu BLE
01:46:01 - Vor 10 Stunden
Hyper 2000 Connection Status geändert zu BLE Fehler
01:33:49 - Vor 10 Stunden
Hyper 2000 Connection Status geändert zu BLE
00:03:49 - Vor 12 Stunden
15. Juli 2025
Hyper 2000 Connection Status geändert zu Zendure Cloud und BLE
14:32:08 - Vor 21 Stunden
Hyper 2000 Connection Status geändert zu BLE
14:32:07 - Vor 21 Stunden
14. Juli 2025
Hyper 2000 Connection Status geändert zu Zendure Cloud und BLE
22:27:20 - Vorgestern
Hyper 2000 Connection Status geändert zu BLE"

Die Integration ist so absolut unbrauchbar...
Jetzt kann ich den Hyper wieder auf Werkseinstellungen zurücksetzen, neu ins WLAN einbinden und komplett neu konfigurieren. 
Und das jedes Mal, wenn dieser Fehler durch die Integration getriggert wird.

Ich hatte die Integration zuletzt über drei Wochen deaktiviert und in diesem Zeitraum keinerlei Probleme aber natürlich auch keine Daten in HA...

Barrage83, 2025-07-16:
keine Ahnung wie das mit Bildern hier läuft....

[image]

man sieht schon die wilde wechslerei... ganz links... und dann nur noch Sekundenbruchteile bei Neustarts etc... läuft....

Barrage83, 2025-07-16:
da du viel von neu ins WLAN einbinden geschrieben hast....  hast den hyper im Router gelöscht? bzw den Shelly aus'm CT gekickt und wieder aktiviert??? das hilft ungemein... wie ich gehört habe muss man das sogar machen wenn man ins hems wechselt