Zendure/Zendure-HA #388: The Hyper 2000 smart mode seems to not always work

nicolinuxfr, 2025-06-15:
I have noticed that sometimes, the smart mode seems to stop working on my Hyper 2000 managed locally through my own MQTT server. For instance here, we can clearly see that the Hyper 2000 power output (blue) do not match the p1 sensor I'm giving to the integration (yellow).

[image]

I'm not sure why, since everything seems to be working fine. Here are some logs where it seems the integration does correct the correct value and transmits it to the battery, but at the same time, the Hyper 2000 seems stuck at an output of 0 currently. Earlier, as you can check on the graph, it was the opposite, the battery was sending electrons while the sensor indicating what was needed was at 0 W.

```
homeassistant  | 2025-06-15 21:44:41.887 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 1668587, 'product': 'solarFlowPro', 'timestamp': 1750016675, 'properties': {'lowTemperature': 0, 'autoHeat': 0, 'heatState': 0, 'invOutputPower': 0, 'ambientLightNess': 20, 'ambientLightColor': 65280, 'ambientLightMode': 0, 'ambientSwitch': 0, 'OldMode': 0, 'pass': 0}}
homeassistant  | 2025-06-15 21:44:42.444 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 366 power: 0 operation: 2
homeassistant  | 2025-06-15 21:44:42.444 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Wait 10 sec for state change p1: 366
homeassistant  | 2025-06-15 21:44:42.444 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 366 power: 0 operation: 2
homeassistant  | 2025-06-15 21:44:42.444 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Wait 10 sec for state change p1: 366
homeassistant  | 2025-06-15 21:44:42.926 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 1668589, 'product': 'solarFlowPro', 'timestamp': 1750016676, 'properties': {'circuitCheckMode': 0, 'chargingTime': 0, 'chargingType': 2, 'dspversion': 8460, 'hyperTmp': 3091, 'strength': 3, 'socStatus': 0, 'gridReverse': 2, 'phaseCheck': 0, 'reverseState': 0}}
homeassistant  | 2025-06-15 21:44:43.968 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 1668591, 'product': 'solarFlowPro', 'timestamp': 1750016677, 'properties': {'outputLimit': 1200, 'inputLimit': 1200, 'acMode': 1, 'gridStandard': 1, 'inverseMaxPower': 1200, 'chargingMode': 2, 'energyPower': 467, 'lampSwitch': 1, 'BatVolt': 0, 'clusterSw': 0}}
homeassistant  | 2025-06-15 21:44:44.997 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 1668593, 'product': 'solarFlowPro', 'timestamp': 1750016678, 'properties': {'autoModel': 8, 'gridOffMode': 0, 'localState': 0, 'ctOff': 1, 'phaseSwitch': 1}}
homeassistant  | 2025-06-15 21:44:48.064 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 1668599, 'product': 'solarFlowPro', 'timestamp': 1750016681, 'properties': {'hyperTmp': 3081}}
homeassistant  | 2025-06-15 21:44:48.810 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/log => {'messageId': 1668601, 'logType': 2, 'timestamp': 1750016682, 'log': {'sn': 'EE1LYMGJM320797', 'params': [35, 4, 355, 365, 362, 357, 4850, 4860, 4860, 4850, 0, 0, 0, 0, 323, 324, 324, 323, 324, 324, 324, 323, 0, 0, 0, 0, 0, 0, 0, 0, 8449, 257, 257, 257, 23, 21, 22, 22, 4545, 5114, 2072, 2135, 4113, 4113, 4113, 4113, 2, 0, 5, 1, 2365, 0, 5, 216, 0, 56, 0, 0, 68, 0, 0, 25, 0, 0, 156, 366, 3, 3, 5, 0, 1200, 0, 0, 8, 1, 2, 0, 0, 4870, 35, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 100, 96, 29, 1355, 255, 0, 0, 100, 10, 0, 4835, 5001, 1, 322, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 0, 1]}}
homeassistant  | 2025-06-15 21:44:56.235 INFO (paho-mqtt-client-nicoflo) [custom_components.zendure_ha.zendurermanager] Topic: Zendure Manager /gDa3tb/Hyper 2000/properties/report => {'messageId': 1668616, 'product': 'solarFlowPro', 'timestamp': 1750016689, 'properties': {'electricLevel': 35}}
homeassistant  | 2025-06-15 21:44:57.413 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 367 power: 0 operation: 2
homeassistant  | 2025-06-15 21:44:57.413 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Start discharging with p1: 367
homeassistant  | 2025-06-15 21:44:57.413 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 367 stateManagerState.DISCHARGING capacity: 0 max: 1200
homeassistant  | 2025-06-15 21:44:57.413 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 367 power: 0 operation: 2
homeassistant  | 2025-06-15 21:44:57.413 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Start discharging with p1: 367
homeassistant  | 2025-06-15 21:44:57.414 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update setpoint: 367 stateManagerState.DISCHARGING capacity: 0 max: 1200
homeassistant  | 2025-06-15 21:45:21.412 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 367 power: 0 operation: 2
homeassistant  | 2025-06-15 21:45:21.412 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Wait 10 sec for state change p1: 367
homeassistant  | 2025-06-15 21:45:21.412 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Update p1: 367 power: 0 operation: 2
homeassistant  | 2025-06-15 21:45:21.412 INFO (MainThread) [custom_components.zendure_ha.zendurermanager] Wait 10 sec for state change p1: 367
```

Is there something I should try ?

harrymayr, 2025-06-16:
Can you change input- and outputLimit to 0 in case you use smart-mode?
In my experience unexpected behavior can occure at some point, if these values are not 0 during smart-mode. I use an automation for that.

nicolinuxfr, 2025-06-16:
> Can you change input- and outputLimit to 0 in case you use smart-mode? In my experience unexpected behavior can occure at some point, if these values are not 0 during smart-mode. I use an automation for that.

Seems counter-intuitive to me but I tried and it seems to work, after like 5 minutes of testing. I will report back if it's OK tomorrow.

Thanks 🙂