Zendure/Zendure-HA #390: Hub1200 sensor.solarflow_gridpower

rony2615, 2025-06-16:
Seid Wochen habe ich immer den gleichen Wert bei sensor.solarflow_gridpower. Neustart hat nichts gebracht. Was kann man machen?

harrymayr, 2025-06-16:
@rony2615 Was erwartest du an diesem Sensor? Meines Wissens steht da im CT-Mode der App, der Wert, der entweder vom verbundenen Shelly 3PM geliefert wird, oder die Summe der Werte der verbundenen Steckdosen.
Die Zendure-HA Integration nutzt aber den angegebenen P1-Sensor und nicht diese Werte, also wird gridPower nicht benötigt und da, wenn du localMqtt verwendest, die Kommunikation zur Cloud sowieso reduziert ist, würde ein Regeln damit auch schwierig.

rony2615, 2025-06-17:
@harrymayr, danke für die Infos ; jedoch geht der gridPower -Integration bei Hub 2000 ,aber nicht beim Hub 1200; der sich ständig aufhängt.