Zendure/Zendure-HA #391: HEMS Update Hyper 2000

peterfieber, 2025-06-16:
Zendure just releases the HEMS Update for Hyper 2000.

https://forum.zendure.com/d/17320

Do you have any ideas if this integration is working still normally after the update? 
They did some major changes also in the lokal and wifi communication.

Lets see if its really better...

C0rd0n, 2025-06-16:
I’ve installed the firmware update for HEMS. The integration is still working as expected. However, I haven’t activated HEMS in the Zendure app yet, as I’m concerned it might interfere unintentionally.

Christian23996, 2025-06-17:
inverseMaxPower is still 1200 W (AC outgoing 1200 W / AC ingoing 1200 W)

thunbergi, 2025-06-17:
> [@thunbergi](https://github.com/thunbergi) If you use HEMS, can you already say how HEMS is going?

HEMS is going well, i had some troubles with the shelly pro3em, was obliges to remove and to add again.

closed 2025-07-22 by FireSon