Zendure/Zendure-HA #395: Keine Daten trotz Verbindung

CaptSonic, 2025-06-17:
Hallo,

seit heute morgen habe ich das Problem, dass ich keine Daten mehr von meinem Hyper2000 über die Integration bekomme.
Habe schon mehrmals neuhinzugefügt und auch einen neuen Account angelegt.

Laut Entitäten schwankt die Verbindung immer zwischen lokalen MQTT und Verbindung mit der App, aber leider füllen sich die anderen Entitäten nicht.

[image]

Aufgetreten ist es nachdem ich heute morgen kurzeitig den Server gewechselt habe, da das neue HEMS nur falsche Daten angezeigt hat und die Zendure App gehangen hat. Habe wieder auf den EU-Server gewechselt, aber leider passiert nichts.

Gibt es irgendwo ein Log, wo man genauer nachgucken kann?

harrymayr, 2025-06-18:
> Habe schon mehrmals neuhinzugefügt und auch einen neuen Account angelegt.

Auch HA neu gestartet? Log gibt es, wenn man Debug  Protokoll aktiviert.

[image]

AlphaZuluNRW, 2025-06-18:
Ich habe vorhin H neu gestartet. Danach tauchte das Problem bei mir auch auf:

[image]

Dadurch funktionoert natürlich jetzt gar nichts mehr und die Akkus werden gerade mit maximaler Leistung entladen ...

harrymayr, 2025-06-19:
Wenn die Geräte in der App auf "Offline" stehen, kann man mit einem Druck auf die IoT Taste des Hypers für ca. 6 Sekunden die Verbindung neu initialisieren. Das hatte ich auch schon mehrfach. Da ist ein weiterer Indikator, dass der Hyper nicht.mehr in der Advertisement Liste von BT auftaucht. 
Wenn @AlphaZuluNRW noch sieht, dass die Batterien mit voller Leistung entladen werden, kann es das eigentlich nicht sein.

AlphaZuluNRW, 2025-06-19:
Jetzt kommen plötzlich wieder Daten. Ob das jetzt an einem Neustart der Fritzbox lag oder Zendure zeitgleich etwas geschraubt hat, weiß ich natürlich nicht...

AlphaZuluNRW, 2025-06-19:
Kann es sein, dass die Chinesen heute Cloudprobleme haben? EcoFlow ist ebenfalls down, so dass auch da weder über die App noch über die entsprechende Integration ein Zugriff möglich ist. Das sieht man auch "schön", wenn man nach "EcoFlow down" googelt. 

Wie gesagt: Zendure läuft hier jetzt wieder perfekt, EcoFlow flackert teilweise, läuft aber definitiv in China nicht sauber ...

AlphaZuluNRW, 2025-06-20:
@harrymayr , leider kommen jetzt wieder keine Daten mehr. Auch die Daten der App sind eingefroren. Die tut zwar so, als ob noch 12 % im Akku sind und entladen wird, aber das stimmt nicht, wie ich am Stromzähler sehen kann.

Das kann doch eigentlich nur bedeuten, dass Zendure ein Problem hat - oder?

AlphaZuluNRW, 2025-06-20:
Jetzt geht's wieder. Das 6er HA-Release habe ich extra noch nicht installiert und BT ist definitiv auf der ToDo-Liste 😊.

olifoss, 2025-06-21:
I've the same problem, get no MQTT data. I'm registered as developer (run curl-script), have appkey and secret. With this credentials an appkey as topic I can log in with MQTT Explorer. Get no data. The Zendure-HA shows that I have 2 devices and 28 entities.  The devices and entities are shown in the dashboard. Zendure HA get's no data for the entities. I've installed Mosquitto Broker in HA and the MQTT-Integration is bound to the Mosquitto MQTT Broker (used for Tasmote smart meter D0 device).

The Zendure SolarFlow is Wifi connectet, no Bluetooth.

CaptSonic, 2025-06-21:
Ich habe eine neue HA-Instanz aufgesetzt um auszuschließen, dass etwas bei meiner Hauptinstanz nicht funktioniert.
Leider keine Änderungen. Anscheinend kommen Daten per MQTT rein, aber werden nicht auf die Entitäten angewendet.

Im Protokoll steht auch, dass der Zendure Manager zu viele Anfragen generiert.

Meine 3 Akkus werden auch nicht erkannt.

AlphaZuluNRW, 2025-06-21:
Hier läuft es jetzt wieder stabil. Daher nehme ich an, dass die Server in China Schluckauf hatten.

Was passiert eigentlich, wenn man zwei Hyper2000 einsetzt? Können die dann beide mit jeweils 1200 W laden und entladen oder steuert das die Integration dann analog zur ZendureCloud, wenn dort ein Cluster gebildet wird?

FireSon, 2025-06-21:
@AlphaZuluNRW Hast du auch ein SF800 oder ein Hyper2000? Und du hast nichts geaendert? Und hast du lokale MQTT oder cloud?

Die device clusters machen genau das, und etwas mehr, weil die Leistung abhangig ist von de relative Kapacitat => electricityLevel - minSoc. 
Es ist sehr wichtig um die richtige device Clusters zu wahlen! Jedes Land hat auch seine eigene Gesetzgebung. Sehe auch die readme...

CaptSonic, 2025-06-21:
Jetzt funktioniert bei mir auch wieder alles:

Der Shelly, der als Bluetooth Proxy dient, hatte nur Passives Scannen an, auf Aktiv gesetzt und alle Daten kommen..

closed 2025-07-22 by FireSon