Zendure/Zendure-HA #397: Move of entities to new Zendure-HA from old MQTT

truffner99, 2025-06-18:
Well, I have to say, the thing has been running extremely well for a few days now. I
t always switches from smart to off, regulates everything via automations, adjusts grid power depending on surplus, etc. 
I'm really impressed! Thank you so much for the great work and your prompt help every time! If everything holds up this well, I'll put something in the donation jar too. :-)

But now I have another problem. I had set up my system on the other solution MQTT, from zmaster/Solarflow and of course other entities with Integralsensor/Riemann in the energy dashboard. How can I change these now without losing my history?
I had already tried simply replacing the existing helpers with the entities from your solution so as not to change the name of the sensor, but that messed everything up. What can I do here?

[image]

[image]

faxenclown120d, 2025-06-18:
Have you ever tried this: 
Schluss mit Datenchaos - So überträgst Du Statistiken in Home Assistant!
https://www.youtube.com/watch?v=c36dj1z78j8

... as your sensor names are "pretty German" i think you won´t have any issues following this tutorial. But be aware it´s at your own risk (i killed a 10d short term statistic somehow, somewhere).
I also started with the zmaster integration and i am now running the 1.0.47 in manual mode with my own scripts and automations (power-out offset, script delay bar, battery dump function over night steering my dehumidifier). I´m quite happy so far!

truffner99, 2025-06-20:
Cool, thanks.  I tested it for some my old entities and works well. Good Tip I will go and use this for the current ones as well, when I have same time.

closed 2025-07-22 by FireSon