Zendure/Zendure-HA #408: 1.1.0-pre3 / Hyper eingefroeren und WLAN offline

Kieft-C, 2025-06-20:
Moin, Moin,

ich bin leider nicht ganz sicher was passiert ist.
ich habe die 1.1.0-pre3 version heute installiert und diese lief auch relativ lang sauber.

Jetzt vor 3/4h stellte ich fest, dass scheinbar teilweise sensoren festhingen und eingefroeren sind.
Dann gab es das Feedback als Verbindungsstatus plötzlich nur noch "Zendure App".

In der Zendure App zeigte er mir auch gar nichts mehr an und nur via Bluetooth gab es eine Verbindung.
Geräteverbindung wird als Offline zum WLAN angezeigt.

Log Datei is nicht easy, da sie bei jedem Neustart von HA automatisch deaktiviert wird.

Any Ideas?

Kieft-C, 2025-06-20:
17:30 bis 22:00 vielleicht?

Ich sehe nun auch nicht mal mehr in HA die Bluetooth Anbindung.

[image]

FireSon, 2025-06-21:
Die Zendure App zeicht immer die letzt verfugbare Information, auch wenn das Gerat seit langere Zeit nicht reagiert.

Linus86529, 2025-06-21:
Ich nutze derzeit noch die Version 1.1.0-pre1 und habe diese „Freeze“-Ereignisse mehrmals hintereinander beobachtet. In der Regel hilft bei mir der „MQTT Rest!”, damit der Hyper2000 wieder weitermacht. In den Kurzeitaufzeichnungen habe ich mir die letzten Tage angeschaut und gesehen, dass ca. alle fünf Stunden ein „Freeze“-Ereignis zum Kommunikationsstopp beim Hyper2000 führt.

Ein Beispiel für den letzten Freeze ist gestern Abend um 21:40 Uhr aufgetreten.

[image]

FireSon, 2025-06-21:
@Linus86529 Wie oft hast du diese freeze? Das ist wichtig fur die Fehlersuche. Hasst du das bei entladen oder auch bei laden?

Kieft-C, 2025-06-21:
Habe Abends den Hyper neu gestartet und die Integration neu eingebunden.
Seit dem (toi, toi, toi) kein Freeze. In 1h sind 24h um und es geht ins klarere "Entladen".
Bis dato war gesamte Tag laden & bypass, ohne jegliche Probleme im Einsatz. Ich berichte weiter.

Ich stoppe einmal jetzt den DeBug und starte ihn neu, dass die datei nicht ewig lang ist.

Kieft-C, 2025-06-21:
@FireSon 
Seit wenigen Tagen ist aber mit nutzung der 1.1.0-Pre versionen die Statistik innerhalb der ZendureApp nicht mehr korrekt.
Ist dies vorab schon so gewesen und nur weil ich nicht lokales MQTT vorab genutzt hab war es mir nie aufgefallen?

bzw. gibt es ne möglichkeit diese wieder zu füttern mit den korrekten Daten innerhalb des lokalen MQTT Einsatzes?

Linus86529, 2025-06-22:
Hi @FireSon, ich bin erst seit heute Mittag wieder zuhause.

Die Freezes treten bevorzugt in den langen Entladephasen über Nacht auf. Die konnte ich am nächsten Morgen immer mit "MQTT Reset" beseitigen. Ein "MQTT Reset" wird vorübergehend von einer Automation erledigt. 

Freezes in der Ladephase habe ich nur einmal am Fr. 20.Juni ab 12:12:50 bis 14:23:39 gefunden. Dieser war nach ca. 2 Std. ohne mein Zutun wieder verschwunden. Ob es von den selbst verschwindenden Freezes noch mehr gab, kann icj leider nicht sagen. 

[image]

Linus86529, 2025-06-22:
@FireSon ich bin mir nicht sicher ob es einen Zusammenhang zwischen der 2 Sekunden Update-Rate gibt, da ich in der v1.0.47 in über längeren Zeitraum diesen Wert in der const.py stehen hatte.

[2 / 2](https://github.com/FireSon/Zendure-HA/issues/258#issuecomment-2939732043) und [1 / 1](https://github.com/FireSon/Zendure-HA/issues/258#issuecomment-2967909172).
Mit beiden Update-Raten hatte ich in der v1.0.47 diese Freezes nicht gesehen.

FireSon, 2025-06-22:
@Linus86529 Vielen Dank fur die Info's! Bevor version 1.1.0 wird die freeze automatisch behoben, und um Mitternacht einmal immer einen reset durchgefuhrt. Deswegen hat nimand einen Problem. Mit version 1.1.0 ist die absicht das Problem letztendlich zu finden 😄 

Die 2 secunden bin ich mir auch nicht sicher, das 'gefuhl' ist das der freeze schneller statt findet...

Kieft-C, 2025-06-25:
Weiterhin laufen noch Dinge ab die ich nicht 100% nachvollziehen kann.

18:03:17 hat sich der connection Status zu "Lokal MQTT und Zendure App" geändert
19:52:46 ohne jegliches zutun auf "Zendure App und BLE"
19:53:08 wieder ohne Eingriff auf "lokaes MQTT, Zendure App und BLE"

Debug war deaktiviert in dieser Zeit leider. Nur Mosquitto Log hab ich

2025-06-24 18:01:54: Client <unknown> disconnected due to protocol error.
[18:02:00] INFO: Successfully send discovery information to Home Assistant.
[18:02:04] INFO: Successfully send service information to the Supervisor.
2025-06-24 18:03:12: New connection from 172.30.32.1:45761 on port 1883.
2025-06-24 18:03:12: New client connected from 172.30.32.1:45761 as <gelöscht> (p2, c1, k60, u'homeassistant').
2025-06-24 18:03:16: New connection from 192.168.1.7:33863 on port 1883.
2025-06-24 18:03:16: New client connected from 192.168.1.7:33863 as homeassistant (p2, c0, k60, u'homeassistant').
2025-06-24 18:31:55: Saving in-memory database to /data//mosquitto.db.
2025-06-24 19:01:56: Saving in-memory database to /data//mosquitto.db.
2025-06-24 19:31:57: Saving in-memory database to /data//mosquitto.db.
2025-06-24 19:52:36: Client KTE<gelöscht> closed its connection.
2025-06-24 19:53:08: New connection from 192.168.1.101:57084 on port 1883.
2025-06-24 19:53:08: New client connected from 192.168.1.101:57084 as KTE<gelöscht> (p1, c1, k60, u'KTEfLQ5L').

und
"Client KTE<gelöscht> closed its connection." kommt öfters in der ganzen Nacht, aber Connection Status bleibt stabil gleich.

Gibt es sonst noch etwas, was ich checken könnte?

Kieft-C, 2025-06-26:
[image]

Selbe Phänomen. Freeze des Hypers.
Communication ändert sich auf: Zendure App und BLE
Dieses mal hab ich Testweise nur Homeassistant Deep neu gestartet und dann ging es wieder,
aber mit Lokal Mqtt und Zendure App ohne BLE.
Dies war das letzte mal 1:1 so, wobei dann irgend wann nach ein paar stunden später plötzlich er BLE wieder fand und communicaiton änderte.

26.Apr - 20:22:48 fand die Änderung der Communication statt.
Mosquitto Log sagt in der Zeit:
2025-06-26 20:03:34: Saving in-memory database to /data//mosquitto.db.
2025-06-26 20:19:14: Client KT<gelöscht> disconnected due to protocol error.
2025-06-26 20:19:14: New connection from 192.168.1.101:63334 on port 1883.
2025-06-26 20:19:14: New client connected from 192.168.1.101:63334 as KT<gelöscht> (p1, c1, k60, u'KTEfLQ5L').
2025-06-26 20:33:35: Saving in-memory database to /data//mosquitto.db.

Supervisor Log:
2025-06-26 20:16:53.803 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
2025-06-26 20:19:14.344 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'KT<gelöscht> '
2025-06-26 20:19:14.805 INFO (MainThread) [supervisor.auth] Successful login for 'KT<gelöscht>'
2025-06-26 21:16:53.813 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state running

Mehr konnt ich nicht finden, BLE Log gibt es nicht, da Debug deaktiviert war.
@FireSon - gibt es etwas, was ich tun kann für besseren Log oder Fehleranlayse?
Debug laufen lassen und zeitstempel mit +/-10min dir hier weiter anzeigen?

Kieft-C, 2025-06-27:
> [@Kieft-C](https://github.com/Kieft-C) Vielen Dank fur deine arbeit!!
> 
> Aber, Heute habe ich verstanden das Zendure das Problem gefunden hat, und das mit einen Firmware update, das einfrieren behoben ist. Ich habe jetzt einen test firmware bekommen, und teste das in die nachste Woche. 😄😄😄

Aufregend! Bin gespannt. Falls ich mit Beta-Testen soll, let me know.

Kieft-C, 2025-06-27:
@FireSon 

Just as information:
Ich hab heute wieder nen freeze gehabt.
Hatte aber als Test eine automation erstellte sie bei Wechsel des Connection Status in “Zendure und BLE” wechselt das adding mosquitto Broker neu startet. Und siehe da. Binnen weniger Sekunden zurück zu “lokal mqtt, Zendure App und BLE” ohne große Probleme.

Was vermutlich dem mqtt reset gleich kommt?

FireSon, 2025-06-28:
@Kieft-C Das wurde bedeuten das Mosquitto irgend ein Problem hat! Ist wichtig um weiter zu beobachten!