Zendure/Zendure-HA #409: 1.1.0 pre3 Local MQTT

thhoppst, 2025-06-21:
Ich habe den Hub 2000 direkt mit dem HA über bluetooth verbunden (kein ESPHome und auch sonst kein Proxy verwendet).
Nach Umstellung bekomme ich die Meldung_

[image]

Aber leider wird nichts über mqtt ausgelesen bzw reingeschrieben?

[image]

Im MQTT Explorer sehe ich das was geschrieben bw gelesen wird:

[image]

Aber wie gesagt alles auf unbekannt bzw keine Daten vorhanden :(

FireSon, 2025-06-21:
You should keep your deviceId's private 😉 

Did you follow the Local mqtt: [instructions](https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt)?
The important questions/answers can be found there...
If all is working properly, and you still do not have values, let me know.

thhoppst, 2025-06-21:
I followed the instructions exactly but unfortunately I don't get any values

FireSon, 2025-07-31:
@thhoppst Can you try the latest version 1.1.0-pre13? The local mqtt should work there as well...

closed 2025-07-31 by FireSon