Zendure/Zendure-HA #410: Feature Request: Wallbox berücksichtigen

AlphaZuluNRW, 2025-06-22:
Ich habe heute ein "Problem" festgestellt. Die Integration reagiert schneller als meine Wallbox beim Überschussladen. Das führt dann dazu:

- Die Wallbox nimmt alles an Strom, was gerade eingespeist wird (bis zum Maximum, was Auto bzw. Box können).
- Kommt weniger Strom vom Dach schaltet die Box zurück und bei nur noch 1,4 kW ab.
- Die KEBA-Box (SolarEdge) ist aber recht träge, schaltet also erst zurück, wenn schon ein paar Sekunden Strom aus dem Netz kommt.
- Die Zendure-Integration sieht aber, dass da Netzstrom gezogen wird und speist ein.
- Das wiederum bekommt die Wallbox mit und schaltet nicht zurück.
- Nach und nach geht die dann auf das Maximum der Zendure-Einspeisuung und leert die Akkus, was ja nicht der Sinn des Überschussladens ist.

Ich stelle mir das so vor, dass man der Integration auch einen Sensor nennen kann, der den Wallbox-Verbrauch "kennt", analog zur Einspeisung/Netzbezug. Wenn dort dann zu sehen ist, dass Strom ins Auto fließt, soll aus den Zendure-Speichern nichts eingespeist werden.

Ließe sich das einrichten? Oder kann man irgendwie über eine Automation das auch so machen?!?

Also: Wenn Auto lädt, dann Zendure Pause

uremmel420, 2025-06-23:
Ich würde das zunächst mit einen Template Sensor versuchen, der [ Stromzähler Leistung - Wallbox Leistung ] berechnet und damit den Zendure ansteuern wie bisher. Damit wird die Wallbox bei der Zendure Regelung ignoriert.

Etwas so: 

template:
  - sensor:
    - name: "Zendure Stromzähler (ohne Wallbox)"
      unique_id: "zendure_stromzaehler_ohne_wallbox"
      device_class: power
      state_class: measurement
      unit_of_measurement: "W"
      state: >
        {% set sz = states('sensor.stromzaehler_leistung') | float(0) %}  # Stromzähler
        {% set wb = states('sensor.wallbox_leistung') | float(0) %}         # Wallbox
        {{ (sz - wb) }}

Ein Problem könnten hierbei die unterschiedlichen Aktualisierungszeiten der beiden Sensoren sein.
Testen kann ich das leider nicht.

Wenn deine PV Anlage weniger Strom bringt, als dein Auto maximal laden kann, geht auch einfach:

template:
  - sensor:
      - name: "Zendure Stromzähler"
        unique_id: "zendure_stromzaehler"
        device_class: power
        state_class: measurement
        unit_of_measurement: "W"
        state: >
          {% set sz = states('sensor.stromzaehler_leistung') | float(0) %}  # Stromzähler
          {% set wb = states('sensor.wallbox_leistung') | float(0) %}         # Wallbox

          {% if wb > 100 %}  # Auto lädt
            0
          {% else %}  # Auto lädt nicht
            {{ sz }}
          {% endif %}

Damit sollte dein Zendure nur Strom einspeisen, wenn das Auto nicht geladen wird.
Ein noch größerer PV Überschuss wird jedoch ignoriert und lädt den Zendure nicht.

harrymayr, 2025-06-23:
> Ließe sich das einrichten? Oder kann man irgendwie über eine Automation das auch so machen?!?
> 
> Also: Wenn Auto lädt, dann Zendure Pause

Ist doch ein prima Einsatzzweg für "Smartes nur Laden". Sobald die Wallbox Strom zieht, den Zendure Manager darauf umstellen.

AlphaZuluNRW, 2025-06-23:
Die Template-Idee setze ich mal um. Die Wallbox liefert zwar die Daten nicht ganz so schnell, wie der Stromzähler, aber Sekundentakt ist das immer noch.

Ich werde berichten, was da passiert :-)

AlphaZuluNRW, 2025-06-23:
Ich habe es jetzt mal so umgesetzt:

Wenn das Auto geladen wird und die Entnahme aus dem Akku mehr als 50 Watt beträgt, dann setzt das Template einfach mal die virtuelle Netzeinspeisung auf 0, sonst nimmt es den realen Wert.

Mal sehen, was dann gleich passiert, wenn ich zuhause bin.

```
{% set ladestrom = (states('sensor.draussen_carport_wallbox_sola_wall_001_ladestrom') | float(default=0)) * 1000 %}
{% set einspeisung = states('sensor.stromspeicher_stro_spei_004_einspeisung_w') | float(default=0) %}
{% set bezug = states('sensor.haus_eg_technikraum_strom_bezug_einspeisung_mess_stro_001') | float(default=0) %}

{% if (ladestrom) > 0 and einspeisung > 50 %}
  0
{% else %}
  {{ bezug }}
{% endif %}

```

AlphaZuluNRW, 2025-06-23:
Als Helfer-Template funktionierte das irgendwie gar nicht, da immer 0 herauskam ...

Ich habe das jetzt noch einmal in der template.yaml gemacht und zwar so:

```
      - name: "Strom: Bezug/Einspeisung für Akkus bei Laden des Autos"
        unique_id: "strom_bezug_einspeisung_fur_akkus_beim_laden_des_autos"
        unit_of_measurement: "W"
        state: >
            {% set ladestrom_auto = (states('sensor.draussen_carport_wallbox_sola_wall_001_ladestrom') | float(0)) * 1000 %}
            {% set batterieeinspeisung = states('sensor.stromspeicher_stro_spei_004_einspeisung_w') | float(0) %}
            {% set bezug = states('sensor.haus_eg_technikraum_strom_bezug_einspeisung_mess_stro_001') | float(0) %}
            
            {% if batterieeinspeisung == 0 %}
              {{ bezug }}
            {% elif batterieeinspeisung > 50 and ladestrom_auto > 0 %}
              {{ -1.0 }}
            {% else %}
              {{ bezug }}
            {% endif %}

```
Jetzt müsste eigentlich dann, wenn mehr als 50 Watt aus dem Akku in Netz gehen und gleichzeitig das Auto lädt, als Wert "-1 Watt" zurückkommen. Sollte da nicht die Integration dann das Einspeisen abbrechen?

AlphaZuluNRW, 2025-06-23:
So, damit scheint es jetzt zu klappen:

```
{% set ladestrom_auto = (states('sensor.draussen_carport_wallbox_sola_wall_001_ladestrom') | float(default=0)) * 1000 %}
{% set batterieeinspeisung = states('sensor.stromspeicher_stro_spei_004_einspeisung_w') | float(default=0) %}
{% set batterieladung = states('sensor.stromspeicher_stro_spei_004_netz_ladung_w') | float(default=0) %}
{% set bezug_einspeisung = states('sensor.haus_eg_technikraum_strom_bezug_einspeisung_mess_stro_001') | float(0) %}

{% if batterieeinspeisung > 10 and ladestrom_auto > 0 %}
  {{ batterieeinspeisung * -1 }}
{% elif ladestrom_auto > 0 and bezug_einspeisung > 0 %}
  {{ -5.0 }}
{% else %}
  {{ bezug_einspeisung }}
{% endif %}
```

FireSon, 2025-07-31:
This is certainly something I will have a look at, once version 1.1.0 is released. Close this issue for now.

closed 2025-07-31 by FireSon