Zendure/Zendure-HA #417: Hub 2000 Hängt sich auf

PennyOttah, 2025-06-25:
Hallo zusammen 
ist irgendwas darüber bekannt das sich der hub sich  aufhängt seit dem letzten update der Integration ? ich habe das Problem das in den letzten tagen grade über nachts sich der hub aufhängt es werden keine daten mehr in  der app so wie in der Integration aktualisiert und Mann kann auch nichts mehr einstellen der ACE läuft völlig normal weiter erst nach einem Harten Neustart also den hub einmal komplett strom tot machen danach geht es wieder.

harrymayr, 2025-06-26:
> Hallo zusammen ist irgendwas darüber bekannt das sich der hub sich aufhängt seit dem letzten update der Integration ? ich habe das Problem das in den letzten tagen grade über nachts sich der hub [...]

Welche Version, 1.0.47 oder die 1.1.0-pre3?
Ist außerdem das HA Update 2025.6.x durchgeführt worden? Seit dem HA Update von 2025.5.x verschwindet mein Hyper immer wieder aus der BT Advertisement Liste um dann Stunden später wieder zu erscheinen. Wenn in dieser Zeit ein Reconnect getriggert wird (und man hat localMqtt ausgewählt) funktioniert die Verbindung nicht. Erst ein Druck auf dem IoT Knopf am Hyper baut die Verbindung neu auf.
Ggf. ist das beim Hub genauso. Falls du localMqtt verwendest, deaktiviere das mal, starte HA neu und beobachte, ob die Verbindung dann immer noch abbricht.

Hardux, 2025-06-27:
hab das gleiche problem mit hyper 2000
hab 4 geräte, alle an eigener Sicherung und gesteuert über zendure Manager.
sporadisch, ohne jede nachvollziebarkeit frieren mir die geräte ein. mal eins, mal 2. mal alle. an manchen tagen kein problem.
bei mir hilft auch nur komplett stromlos machen. danach läuft direkt wieder
laut GPT Deep research haben auch andere zendure Nutzer mit dem Cluster Probleme, das wird manchmal nicht mehr erkannt.

ZENDURE INTEGRATION v1.0.47 über HACS
HA Core  2025.5.3

mod42, 2025-06-28:
Das "hängen" der App kenne ich auch mit dem Hyper 2000 ohne dass eine Integration mit HA verwendet wird. Ich bin per Zufall auf Eure Unterhaltung gestossen. Das System "kommt" nur wieder, wenn man den Hyper neu startet.

Hardux, 2025-06-28:
mal unabhängig davon ob der bug gefixt wird. das problem könnte man schon easy entschärfen wenn ein softwareseitiger Neustart der geräte möglich wäre....
aber ein. Stecker aus der Steckdose, pv Strings ab, hyper vom Akku heben.
das ist so nervig. ich bin ja immer froh wenn es dunkel ist und das problem auftritt. dann kann ich wenigstens die MC4 Stecker hängen lassen 😂

Christian23996, 2025-06-29:
@Hardux 
Wie hast du das mit den Automationen gelöst? Habe das Problem auch und versuche anhand verschiedener Indikatoren das irgendwie zu monitoren. Die "stillen" Ausfälle sind echt grausam. Bin hier noch in der Findung.

Ich habe drei Hyper 2000 und habe mir einen Helfer (Kombinationssensor aus den Batterie Level) erstellt. Wenn der um mehr als 2 abweicht wird das getriggert, Info auf mein Handy und lokal MQTT Reset. Manchmal reicht der Mqtt Reset allein nicht aus, dann muss ich auch noch zusätzlich den Zendure Manger ausschalten und wieder auf smarte Leistungsregulierung einschalten.

Wie hast du das gelöst?

Hardux, 2025-06-29:
schon bei 2 Prozent Abweichung? das würde bei mir ja dauerhaft auslösen 😂
ich hab 10 Prozent eingestellt. selbst da bekomm ich ab und zu fehlerhafte pushups.
liegt aber auch daran das unser 4ter hyper anderen Modulaufbau hat und wir Überschuss einer bestehenden anlage laden.
dadurch ist der öfters mit den Laderichtungen am umschalten und es kommt schneller zu Differenzen.

Zwei weitere Indikatoren sind output_pack_power und output_home_power.
ich habe 2 input_number die zentral die Ladegrenzen aller hyper einstellt.
die Automation überprüft ob Netzbezug da ist. wird zuviel aus dem netz gezogen obwohl noch leistung von output_home möglich wäre und der akkuzustand das auch zulässt, dann gibt es eine pushup. das gleiche dann auch wenn zu viel verkauft wird, obwohl noch mehr Ladeleistung möglich wäre. in meinem fall bei 4 hyper ist das 4800W Ausgangsleistung und 6400W Ladeleistung
für die Sonderfälle, sobald ein Akku voll ist, setzt die eine, oder ein Akku leer, setzt die andere Automation aus. dafür vergleiche ich die akkustände mit meiner entsprechenden input_number

diese Automationen funktionieren bei mir zuverlässig, auch mit unserer bestehenden anlage.
ich überwache auch sensoren ob sie auf unknown oder so gehen. aber das funktioniert nicht, da die werte alle einfrieren aber trotzdem verfügbar sind.
in den fällen ist mir auch aufgefallen, wenn ich die Integration neu starte, die akkustände werden mir direkt wieder angezeigt, alles andere bleibt auf 0. allerdings stimmt der wert trotzdem nicht 😂

Christian23996, 2025-06-29:
@Hardux 
Warum muss deine Ladegrenze so oft variieren? Ich lasse die den ganzen SolarSommer auf min10/max100 stehen. Nur zum Winter hin ändere ich die. Dafür lohnt sich bei mir keine Automatik.

Werden deinen Ansatz mal verfolgen und schauen ob's bei mir ggf. auch funktioniert. 
Als Ladegrenze würde ich dann den Wert SOC Maximum / SOC Minimum übernehmen.

Ja. leider funktioniert "unknown" / "unavailable" nicht.

Hardux, 2025-06-29:
ich ändere auch nicht so oft. Sommer 15/90 Winter 30/90
aber so habe ich einen zentralen Regler um alle 4 anzupassen. ich verwende dafür eine ränge slider card. damit kann ich 2 Regler auf einen Balken legen und habs im dashboard einfach anzupassen.
ja kannst du natürlich auch direkt number.hyper_1_soc_set nehmen.
wie sieht deine logik für die Ladestände aus?
ich hab die durchschnitte von den drei anderen und vergleiche mit dem 4.

```python
- alias: Hyper – Akkuabweichung
  mode: single
  trigger:
    - platform: time_pattern
      minutes: "/5"
  variables:
    a1: "{{ states('sensor.hyper_1_electric_level') | float(0) }}"
    a2: "{{ states('sensor.hyper_2_electric_level') | float(0) }}"
    a3: "{{ states('sensor.hyper_3_electric_level') | float(0) }}"
    a4: "{{ states('sensor.hyper_4_electric_level') | float(0) }}"
    avg_1: "{{ (a2 + a3 + a4) / 3 }}"
    avg_2: "{{ (a1 + a3 + a4) / 3 }}"
    avg_3: "{{ (a1 + a2 + a4) / 3 }}"
    avg_4: "{{ (a1 + a2 + a3) / 3 }}"
    d1: "{{ (a1 - avg_1) | abs }}"
    d2: "{{ (a2 - avg_2) | abs }}"
    d3: "{{ (a3 - avg_3) | abs }}"
    d4: "{{ (a4 - avg_4) | abs }}"
    diffs: >
      {{ [
        {'id': 1, 'value': d1},
        {'id': 2, 'value': d2},
        {'id': 3, 'value': d3},
        {'id': 4, 'value': d4}
      ] }}
    biggest: >
      {{ diffs | sort(attribute='value', reverse=true) | first }}
  condition:
    - condition: template
      value_template: "{{ biggest.value > 10 }}"
  action:
    - service: notify.marcel
      data:
        title: "Zendure Integration"
        message: "⚠️ Hyper {{ biggest.id }} Akkuabweichung {{ biggest.value | int }}% erkannt!"
    - delay: "01:00:00"
```

Hardux, 2025-06-30:
@Christian23996 
das sollte auch funktionieren.

```python
- alias: Hyper – Verbindung weg
  mode: single
  trigger:
    - platform: state
      entity_id:
        - sensor.hyper_1_connection_status
        - sensor.hyper_2_connection_status
        - sensor.hyper_3_connection_status
        - sensor.hyper_4_connection_status
  condition:
    - condition: template
      value_template: >
        {{ trigger.to_state.state | string in ['unavailable', 'unknown', '0'] }}
  action:
    - service: notify.marcel
      data:
        title: "Zendure Integration"
        message: "⚠️ Hyper {{ trigger.entity_id.split('.')[1].split('_')[1] }} Verbindung verloren."
```

Hardux, 2025-06-30:
@Christian23996 
Für mich hat der connectionstatus heute zuverlässig funktioniert. Schau mal bei Dev->Zustände, was er macht. Bei mir geht er auf 0.  Und da der trigger state ist, sollte das gehen. Unknown oder unavailable hat bei mir nicht funktioniert, da der sensor nicht auf unbekannt springt. Seitdem ich zustand 0 hinzugefügt hab, passt es.

Aber deswegen habe ich verschiedene Mechanismen. Es kann ja auch mal anderweitig ein Problem auftreten. Sicherung rausgeflogen oder sonst was. Gibt nix schlimmeres wenn die Anlage nix schafft und man es wochenlang nicht mitbekommt.

Christian23996, 2025-06-30:
Zum Zustand Connection Status hat harrymayr mal folgendes gepostet.

The following is possible:

"name": "Connection Status",
"state": {
"0": "Unbekannt",
"1": "BLE",
"2": "lokal MQTT",
"3": "lokal MQTT und BLE",
"4": "Zendure Cloud",
"5": "Zendure Cloud und BLE",
"9": "Zendure App und BLE",
"10": "Lokal Mqtt und Zendure App",
"11": "Lolal Mqtt, Zendure App und BLE",
"16": "BLE Fehler"

Hab mal im Verlauf meiner drei Hyper "Connection Status" Sensoren geschaut, da könnte es durchaus einen Zusammenhang geben.
Unmittelbar auf den Status 0 folgt dann Status 2, meist noch in der selben Sekunde. 

[image]

Hardux, 2025-07-01:
Funktioniert das, einfrieren Per mqtt reset wieder lauffähig zu machen immer? Ich muss die hyper immer komplett stromlos machen. Damit die wieder laufen. 
Wenn das der Fall ist muss ich mal schauen wie das bei der zendure Integration läuft.

Ich habe schon mal über mosquito mqtt gemacht ohne die zendure integration. aber darüber war keine Steuerung möglich.

Christian23996, 2025-07-01:
Ja, funktioniert bis dato zuverlässig mit dem Mqtt Reset Schalter. 
Lasse die Schalter im Abstand von 15 s nacheinander durchlaufen und zusätzlich noch einmal den Zendure Manager auf aus und anschließend wieder auf Smarte Regelung. Das läuft so seit 5 Tagen bereits problemlos durch. 

Es wird zwar wahrscheinlich mangels eines geeigneten Triggers viel zu oft ausgelöst. Da es augenscheinlich aber der Regelung nicht weiter stört, lasse ich es erstmal so. Ggf kann man hier auch einfach einen stündlichen Auslöser hinzufügen oder noch engmaschiger.

[image]

PennyOttah, 2025-07-02:
Moin
also mein hub hat sich letzte Nacht wieder aufgehängt ca 00uhr hatte heute morgen auch vieles probiert mqtt rest Schalter hat nix gebracht ich musste den hub wieder Komplet Strom tot machen danach ging es wieder , das komische alle daten vom ace so von den Akkus kommen weiter durch dieses Problem beschränkt sich nur auf den hub ich hatte das Gefühl das die Probleme seit dem letzte update angefangen haben also von der hier habe die Version 1.47

PennyOttah, 2025-07-02:
was ich grade noch gesehen habe mein hub wechselt alle paar stunden immer zwischen zendure cloud und ble  auf ble und wieder zurück auf zendure cloud mein ace macht das nicht

Christian23996, 2025-07-02:
@PennyOttah 
In dieser Nacht hatte ich keine Probleme mit meinen Hyper2000, in der Nacht davor schon. Das hat dann zum Glück komplett meine Automation "weggebügelt".
Ich habe bei mir festgestellt, dass ein Mqtt Reset nicht immer ausreicht, in manchen Fällen muss ich dann den Zendure Manager kurz ausschalten, 5-10 s warten und dann wieder auf smarte Regelung stellen.

Ich bin mir nicht ganz sicher, aber wenn man über die Zendure Cloud verbunden ist und nicht lokal über mqtt, ob der Mqtt Reset Schalter dann überhaupt eine Wirkung hat.

Hardux, 2025-07-19:
@Christian23996 
ich hab seit 3 tagen kein Ausfall mehr. Zufall, oder wie sieht es bei dir aus?
bin zur zeit noch nicht über mqtt verbunden, lediglich über zendure cloud.

Christian23996, 2025-07-20:
Habe gerade mal geschaut, heute Nacht hatte ich drei Resets. 3:48 Uhr/ 4:04 Uhr und 5:47 Uhr. Vor ein paar Tagen habe ich noch ein Firmware Update von Zendure installiert, hatte gehofft, dass es dadurch stabiler wird. Kann nicht wirklich sagen das es besser geworden ist.

Christian23996, 2025-07-22:
@Hardux 
Du hast eigentlich nur zwei Voraussetzungen für "local mqtt".
1. Du benötigst einen Bluetooth-Adapter inkl. der HA Bluetooth Integration (Adapter muss in stabiler Reichweite der Zenduregeräte sein)
2. mosquitto muss als MQTT Broker in HA funktionieren

Mehr braucht es nicht! 
Den Rest macht die Zendure-Integration für dich. Die benötigten Benutzer für mosquitto werden alle (pro Gerät einer) automatisch angelegt.
"Local mqtt" anhaken und nach dem HA-Neustart dann nur eine wenig Geduld haben (mind. 10-20 min).

Unter dem Reiter "WiKi"(oben) findest du noch etwas mehr Infos dazu. 
https://github.com/FireSon/Zendure-HA/wiki/Local-Mqtt

Hardux, 2025-07-27:
@Christian23996 

Jou läuft👍🏻
musste noch BT proxys einrichten da der raspi es nicht so weit geschafft hat. einmal MQTT Reset drücken und hat sich wieder connected👌🏻
was ein Segen nicht mehr in die Garage rennen zu müssen 😁

so wie ich das bei mir sehe hab ich keine Probleme das richtig auszulösen. wenn die werte eingefroren sind geht der connection Status auf Zendure App und BLE "9"
ansonsten ist er immer auf Lokal Mqtt, Zendure App und BLE "11"

ja ab und zu wechselt er mal für paar Sekunden seinen Status. aber das kannst du bisschen Puffern damit das die Bedingungen 5 minuten oder so gültig sein müssen.

ich teste das jetzt mal so 
```python

- alias: Hyper – Verbindung weg
  mode: single
  trigger:
    - platform: state
      entity_id:
        - sensor.hyper_1_connection_status
        - sensor.hyper_2_connection_status
        - sensor.hyper_3_connection_status
        - sensor.hyper_4_connection_status
      for: "00:03:00"
  condition:
    - condition: template
      value_template: >
        {{ trigger.to_state.state | string in ['unavailable', 'unknown', '0', '9'] }}
  action:
    - service: notify.zendure
      data:
        title: "Zendure Integration"
        message: "⚠️ Hyper {{ trigger.entity_id.split('.')[1].split('_')[1] }} Verbindung verloren."

    - service: switch.turn_on
      target:
        entity_id:
          - switch.hyper_1_mqtt_reset
          - switch.hyper_2_mqtt_reset
          - switch.hyper_3_mqtt_reset
          - switch.hyper_4_mqtt_reset
          
    - service: notify.zendure
      data:
        title: "Zendure Integration"
        message: "⚠️ Reset durchgeführt"

```

Christian23996, 2025-07-27:
@Hardux 
Genau so habe ich es aktuell auch gelöst. Zusätzlich dazu deaktiviere/aktiviere ich den Zendure Manager (15s aus dann wieder an) noch kurz. Nicht immer reicht der Mqtt- Reset. 

Mal schauen wie die Integration sich noch entwickelt. Mit der aktuellsten Pre-Version ist derzeit kein lokales MQTT mehr möglich, nur Cloud. Ich hoffe, das kommt wieder zurück.

Hardux, 2025-07-27:
@Christian23996 
jou hatte eben auch mal ein Reset gemacht als alles lief. hab ein Skript gemacht der alle 4 resettet bei Button press. dabei ist mir der eine eingefroren. hab ich dann nochmal Reset gemacht, dann ging's.
hab auch nochmal bissi geändert.

so wird nur der resettet der getriggert hat. dazu noch mode auf queued falls 2 gleichzeitig triggern.

```python
    - service: switch.turn_on
      target:
        entity_id: >
          switch.hyper_{{ trigger.entity_id.split('_')[1] }}_mqtt_reset
```

kaluxx, 2025-08-01:
**Device Unavailable in Home Assistant - gelöst durch (widerholtes) Teilen des H2000 and den HA-Zendure App user.**

Bei mr haben die Reset's des H2000 nicht geholfen. Es gab im Log dann den Hinweis:

`Connected to https://app.zendure.tech/eu => Mqtt: mqtteu.zen-iot.com`
`Getting device list ...`
`Found: 0 devices`

Ich habe dann in der App das Gerät nochmals mit dem User geteilt, den ich in HA verwende. -> OK