Zendure/Zendure-HA #418: Zendure Hyper 2000 crash at 0:00

tiimjcb, 2025-06-26:
Hello,
My Hyper 2000 seems to completely crash at 00:00, and it seems to happen only when this HA Integration is enabled.

The battery is stable without the integration.

[image]

If you take a look at this screenshot, the battery isn't communicating anymore at 00:00. And it's not only about communication, in fact, it's not working anymore until I manually restart it (on the screenshot, restarted at 12:15).

[image]

From what I can see on the "Connection status" sensor, when it goes down, it alternates between "Unknown/BLE" and "BLE Error".

Why is this error happening at 00:00 specifically? Maybe it is because of the local communication -- is it possible to force a cloud connection?

In the logs, I have something like that :

```
2025-06-26 00:20:31.618 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 00:32:52.619 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 00:45:08.619 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 00:57:29.618 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 01:09:50.618 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 01:22:06.618 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 01:30:09.548 WARNING (MainThread) [aioairzone_cloud.websockets] WS[60f6ddbfff517e97263683b5]: reconnecting...
2025-06-26 01:34:27.618 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 01:46:48.619 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 01:52:54.423 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] BLE error: 
2025-06-26 01:52:54.426 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 304, in bleMqtt
    await client.disconnect()
  File "/usr/local/lib/python3.13/site-packages/habluetooth/wrappers.py", line 397, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak/backends/bluezdbus/client.py", line 409, in disconnect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/dbus_fast/aio/message_bus.py", line 398, in call
    await future
  File "src/dbus_fast/aio/message_reader.py", line 23, in dbus_fast.aio.message_reader._message_reader
  File "src/dbus_fast/_private/unmarshaller.py", line 868, in dbus_fast._private.unmarshaller.Unmarshaller._unmarshall
  File "src/dbus_fast/_private/unmarshaller.py", line 732, in dbus_fast._private.unmarshaller.Unmarshaller._read_header
  File "src/dbus_fast/_private/unmarshaller.py", line 453, in dbus_fast._private.unmarshaller.Unmarshaller._read_to_pos
  File "src/dbus_fast/_private/unmarshaller.py", line 393, in dbus_fast._private.unmarshaller.Unmarshaller._read_sock_with_fds
EOFError
2025-06-26 01:59:04.620 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 02:11:27.618 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 02:23:48.617 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 02:36:04.618 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 02:48:26.617 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 03:00:45.395 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] BLE error: 
2025-06-26 03:00:45.398 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 304, in bleMqtt
    await client.disconnect()
  File "/usr/local/lib/python3.13/site-packages/habluetooth/wrappers.py", line 397, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak/backends/bluezdbus/client.py", line 409, in disconnect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/dbus_fast/aio/message_bus.py", line 398, in call
    await future
  File "src/dbus_fast/aio/message_reader.py", line 23, in dbus_fast.aio.message_reader._message_reader
  File "src/dbus_fast/_private/unmarshaller.py", line 868, in dbus_fast._private.unmarshaller.Unmarshaller._unmarshall
[... 49 lines cut ...]
2025-06-26 11:49:48.375 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Traceback (most recent call last):
  File "/config/custom_components/zendure_ha/zenduredevice.py", line 304, in bleMqtt
    await client.disconnect()
  File "/usr/local/lib/python3.13/site-packages/habluetooth/wrappers.py", line 397, in disconnect
    return await self._backend.disconnect()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bleak/backends/bluezdbus/client.py", line 409, in disconnect
    reply = await self._bus.call(
            ^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/dbus_fast/aio/message_bus.py", line 398, in call
    await future
  File "src/dbus_fast/aio/message_reader.py", line 23, in dbus_fast.aio.message_reader._message_reader
  File "src/dbus_fast/_private/unmarshaller.py", line 868, in dbus_fast._private.unmarshaller.Unmarshaller._unmarshall
  File "src/dbus_fast/_private/unmarshaller.py", line 732, in dbus_fast._private.unmarshaller.Unmarshaller._read_header
  File "src/dbus_fast/_private/unmarshaller.py", line 453, in dbus_fast._private.unmarshaller.Unmarshaller._read_to_pos
  File "src/dbus_fast/_private/unmarshaller.py", line 393, in dbus_fast._private.unmarshaller.Unmarshaller._read_sock_with_fds
EOFError
2025-06-26 11:55:58.618 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
2025-06-26 12:08:20.617 ERROR (MainThread) [custom_components.zendure_ha.zenduredevice] Timeout when trying to connect to Hyper 2000 ZenE1_73
```

Available if more informations or logs needed

FireSon, 2025-06-28:
@tiimjcb You are probably using version 1.0.47? This version trys to reset the Hyper2000 at midnight. This reset uses Bluetooth and  apperantly your BLE adapter is unable to perform the reset, which causes these errors. You can use the MqttReset switch to do the reset manually. If the BLE signal is not stable enough, this problem will persist!
Fortunately, Zendure has identified the cause of the communication issue, which will be fixed in the next firmware update. After that the reset will not be necessary anymore 😄

FireSon, 2025-06-28:
In your case, I would say do not use local MQTT until the firmware is out!

tiimjcb, 2025-06-28:
> In your case, I would say do not use local MQTT until the firmware is out!

How can I disable this?

FireSon, 2025-06-28:
Use reconfigure:

[image]

And uncheck Use local MQTT

[image]

tiimjcb, 2025-06-28:
thanks, but the MQTT communication was already off. However, I noticed that no SSID and Password were provided : this might explain why after each reset, I had to reconfigure the wifi connection of the zendure, and also, why it failed.

 I tried turning ON the MQTT Connection, and setting up the credentials. Thanks again for your help.