Zendure/Zendure-HA #421: Hyper 2000 verbindung bricht regelmäßig ab

peterfieber, 2025-06-28:
Hallo zusammen,

wahrscheinlich kennt ihr den Fehler schon, aber ich hab aktuell keine Lösung dafür.
Folgender Ablauf passiert bei mir immer wieder:

Verbindung funktionert : BLE, MQTT, Cloud
Dann wechselt es irgendwann auf: BLE, Cloud - keine akutellen Daten mehr
Dann kurze Zeit später auf: 24
Dann auf 26

Ab BLE, Cloud passiert nichts mehr und es werden keine neuen Daten im HA verarbeitet. Neu starten, neu laden führt idr zu BLE Fehler oder nur zu BLE Verbindung und 5 min später dann zu BLE Fehler.

Gibts da ne Lösung für?

Danke und LG

Wunni73, 2025-06-28:
Habe ich auch 
Am Hyper neu die WLAN Verbindung herstellen. (3 sec die Taste drücken, grünes lauflicht)
und dann warten.

Wunni73, 2025-06-28:
Bin auf version 1.0.47 zurückgegangen. Dann hat er ohne warten die Verbindung wieder hergestellt.

FireSon, 2025-06-28:
The version 1.1.0 has no automatic 'repair' in order to check if the modification in the communication with the Hyper2000 are an improvement. In the mean time, Zendure has identified the cause of the communication problem and will fix it in the next firmware update.
The version 1.0.47 should work properly.

closed 2025-06-28 by FireSon