Zendure/Zendure-HA #425: Unknown batteries ??

Psyle80, 2025-06-28:
Ich habe am 26. und 28.6. wohl Akku Kinder bekommen. Keine Ahnung was das für Seriennummern sind, auf jeden Fall nicht meine. Kann sich das jemand erklären? 

I think I had battery-powered babies on June 26th and 28th. I have no idea what those serial numbers are, but they're definitely not mine. Can anyone explain this?

[image]

harrymayr, 2025-06-29:
> Ich habe am 26. und 28.6. wohl Akku Kinder bekommen. Keine Ahnung was das für Seriennummern sind, auf jeden Fall nicht meine. Kann sich das jemand erklären

Du bist nicht der Erste. Was für ein Device hast du, einen Hyper oder einen Hub 2000?

In der 1.1.0-pre3 sollte nach 5 Minuten keine neue Batterie hinzugefügt werden. Welche Version verwendest du?

harrymayr, 2025-06-29:
Es gibt (min.) 2 Möglichkeiten: 
1. Du nutzt localMqtt und in der Nachbarschaft gibt es auch jemanden der  ein Zendure-Device hat (ist noch nicht bestätigt, aber eine Arbeitshypothese)
2. Zendure schickt manchmal falsche Daten über die Cloud. 

Wie gesagt, bei der aktuellen 1.1.0-pre3 sollte die Wahrscheinlichkeit niedriger sein, dass sich fremde Akkus als Devices eintragen.

Psyle80, 2025-07-01:
Ich nutzen kein lokal mqtt - alles nur cloud - muss also von daher kommen, naja stören ja nicht wirklich und wenn ich sie aktiviere kommen eh keine Daten .. merkwürdige ist es aber dennoch 
Danke für eure Antworten