Zendure/Zendure-HA #426: small display error

judie07, 2025-06-29:
with version 1.1.0-pre3 there are small display errors for me.

A very big THANK YOU for the great work.

[image]

fodhamburg, 2025-06-29:
I am on 1.1.0-pre3 as well and have neither of those behaviours.

MacdiverOne, 2025-06-29:
@fodhamburg: +1

FireSon, 2025-06-29:
It probably depends on the devicetype, I think @judie07 you have a SF800Pro? 
I have added the entities to the SF800Pro, so in the next release it should be ok 😄

judie07, 2025-06-30:
Looks good!

[image]

Is it correct that the battery only has 5,100 volts or should it be 50 volts?

[image]