Zendure/Zendure-HA #438: Zendure 3CT Integration

MrFuellort, 2025-07-03:
Hallo, ich habe einen Zendure 3CT Smart Meter und würde diesen auch gern mit in Home Assistant integrieren um so optimal den  Verbrauch mit zu Tracken. Vielen Dank schon mal vorab und auch für deine tolle Arbeit. Die Integration ist grossartig.

harrymayr, 2025-07-04:
Prinzipiell wahrscheinlich machbar, aber solange wir so ein Gerät nicht haben, sind wir auf die Hilfe der Besitzer angewiesen die schon so ein Teil haben ;-).
Wird allerdings ein wenig hin- und hergehen müssen. Der erste Schritt wäre die Info, ob und wie sich das Gerät meldet. Ich habe bei einem Log eines Solarflow 2400AC gesehen, dass sich ein Shelly 3EM mit den folgenden Zeilen meldet.
`[custom_components.zendure_ha.zendurermanager] Adding device: <<seriennummer>> ShellyPro3EM` und
`[custom_components.zendure_ha.zendurermanager] Data: {'id': 193702, 'deviceKey': '<<seriennummer>>', 'snNumber': '<<seriennummer>>', 'name': 'Shelly 3EM', 'productId': 55, 'productKey': '<<productKey>>', 'onlineFlag': '00602', 'productName': 'ShellyPro3EM ', 'wifiStatus': False, 'blueState': False, 'fourGStatus': False, 'isShareFlag': '00602', 'input': False, 'output': False, 'electricity': None, 'hasPack': False, 'restState': 0, 'upsMode': False, 'upgradeStatusDes': '', 'productType': 12, 'deviceType': 1, 'upgradeStatus': {}, 'bindId': 0, 'bindStatus': 0, 'batteryCode': '', 'packList': [], 'inputPower': 0, 'outputPower': 0, 'slowChargePower': 0, 'temperature': 0, 'temperatureUnit': 0, 'remainOutTime': 0, 'bindType': 0, 'seriesMode': 0, 'parallelMode': 0, 'networkType': 0, 'standard': 'c', 'isSwitch': False, 'url': 'https://control.shelly.cloud/images/device_images/SPEM-003CEBEU.png', 'remainOil': 0, 'genMode': 0, 'shortCode': '', 'phaseCheck': 0, 'pass': 0, 'clusterList': [], 'clusterNotice': False, 'nodeStatus': 0, 'clusterId': None, 'phase': None, 'reverseState': 0, 'meterId': 0, 'hemsId': 7830, 'totalPower': 0, 'keyFlag': 0}`

Wenn du also das Debug-log mal anschmeißt, HA neu startest und in dem Log nach den folgenden Zeilen suchst, sollte darunter was ähnliches wie oben für den 3CT erscheinen. Das wäre die erste Info, die wir brauchen. Danach könnten wir eine Version erstellen, die den 3CT einbindet, aber erst mal nicht mit allen Werten schön formatiert.

[image]

DerUnwissende, 2025-07-08:
Moin, daran bin ich auch sehr interessiert und würde in den kommenden Tagen mal schauen, ob ich sowas mit 3ct in den Logs finden kann. Melde mich dann

MrFuellort, 2025-07-23:
[harrymayr] danke für deinen Tipp, ich hab das mal gemacht und den Debüt Log aktiviert. Tatsächlich steht dort auch was vom 3CT api key Seriennummer etc. aber leider steht dann auch not supported. 

Es wird wohl schon dran gearbeitet und einige die im Betaprogramm sind dürfen den schon nutzen und testen. 
Was aktuell bedeutet weiterhin warten bis zum release. Ich danke dir dennoch.

closed 2025-08-28 by github-actions[bot]