Zendure/Zendure-HA #446: Go to Profile > Device Sharing and setup a share for your 2nd account

DerUnwissende, 2025-07-06:
Hej,
as you described in your main Configuration and your hint for creating a second Account to avoid being logged out of the app, i tried to follow your hint.
I created a second Account on Zendure, but when i log in to my first account to share the device i can't add my second account as there is no possibility to add something?
What's my fault? Did i oversee something?

[image]

I would be very pleased about a feedback from you.

Regards
Rolf

Zanooon97, 2025-07-06:
https://github.com/user-attachments/assets/1c25a78b-dfc6-4406-ad53-6640273ff03a

DerUnwissende, 2025-07-07:
@b703d0b4ecc17cd85c8f904cbbe91015 Thanks, you are right... I use a SolarFlow 2400AC...
Thanks for your time and answer..

DerUnwissende, 2025-07-07:
> Screenrecorder-2025-07-06-20-03-02-287.mp4

Thanks for your video... I will try it, like you described..

DerUnwissende, 2025-07-07:
> Screenrecorder-2025-07-06-20-03-02-287.mp4

but a short other question... With which Tool/App did you capture your video? Maybe it' helpful for me to advise problems or solutions to other...

DerUnwissende, 2025-07-07:
> the original screenrecorder from android

ahhh... shit... looks very good, but i am on iOS.... 
Thanks for your fast answer...

Rolf