Zendure/Zendure-HA #449: Error using local mqtt

thevinz62, 2025-07-08:
Hi, 

I try to use the intégration with a local mqtt on my HA.
I have the MQTT Add-on install and it works (i use it for another intégration and it works)

[image]

I have the Bleutooth ok and I see the 2 hypers:

[image]

But even after a long time, all my entities stay to unknown

[image]

Can you help me ?

harrymayr, 2025-07-08:
Did you restart HA after setup or selecting localMQTT?

thevinz62, 2025-07-08:
Yes of course ! 
The version of the intégration is 

[image]

Question, should I see a new topic in MQTT ? I have seen a IOT topic arrived looking in MQTT explorer ?

Thank you

FireSon, 2025-07-31:
The local mqtt is now also added to the latest version 1.1.0-pre13 Please try this version, if the problems remain please reopen this issue.

closed 2025-07-31 by FireSon