Zendure/Zendure-HA #453: SolarFlow 800 Pro - why are the solarPower sensors 3 and 4 binary ? Doesn't make sense to me ...

Gyvate108, 2025-07-09:
The solarPower1 and solarPower2 sensors show the input wattage as W - whereas solarPower3 and 4 are configured as binary sensors, which doesn't make sense to me. Where can this be changed/corrected ?

harrymayr, 2025-07-09:
There is already a change done, but not a Release for now. If you want it right now, cooy the file for the solarhub800 Pro from the devices folder to your custom_components folder of your HA installation and restart HA

harrymayr, 2025-07-10:
Maybe you don't use Version 1.1.0-pre.
Use the version you want and add the lines in solarflow800pro.py with solarPower3 and solarPower4. To do that, just copy the lines with 1&2 and change it to 3&4.
Be sure to keep the indent.

[image]