Zendure/Zendure-HA #455: SF800 Pro & AC2400 - Entitäten Namen & Umrechnung bei Temperatur

Kieft-C, 2025-07-10:
Integration Version 1.0.47

SF800 Pro
Temperatur wird als Hyper temperatur angezeigt als Name & Umrechnung mit Kelvin Nullpunkt ist nicht erfolgt
Derzeit heißt er dort: sensor.solarflow_800_pro_hyper_tmp

PV3 & PV4 Eingang kommt in einer späteren Pre-release version als fix, oder?
in Version 1.1.0-pre3 finde ich es nicht in der solarflow800pro.py datei
Hier sind nur PV1+2 drin. Hatte von @harrymayr ein kommentar gelesen, dass es im 1.1.0-pre3 gefixt sein sollte?

AC2400 selbe Spiel mit Temp. entität und fehlender umrechnung.
mit: sensor.solarflow_2400_ac_hyper_tmp

Kieft-C, 2025-07-10:
@harrymayr 

SF809Pro & AC2400 - Temperatur Berechnung und namen anpassung is aber nicht dort drin, so lese ich es aus der codeline heraus.

harrymayr, 2025-07-10:
Da ist auch noch der PR #451 mit den restlichen Änderungen

harrymayr, 2025-07-24:
Sollte mit der 1.1.0-pre7 jetzt auch als Release verfügbar sein.

closed 2025-07-24 by harrymayr