Zendure/Zendure-HA #457: Workflow Frage - No Issue - Hyper2000 Basis Smartmatch & SF800 für Spitzen Abfang

Kieft-C, 2025-07-10:
Hypothetische frage zu Idee eines Setups bzw. Umsetzung mit Automationen in HA.

Aufbau vom Setup wäre:
Hyper2000 mit 1200W Leistung und 2x AB2000
SF800 mit 800W Leistung und 1x AB1000

Integration ohne Cluster im Smartmatch Mode mit P1 Sensor mittels Shelly 3EM Pro
Automation I
= wenn Hyper2000 Ausgabe an Wohnung = 1200W für 30sec wechsle SF800 in "Cluster on" (oÄ. wie es heißt)
= wenn Hyper2000 Ausgabe an Wohnung unter 1200W für 30sec wechsle SF800 aus "Cluster Off"

Ziel soll es sein, dass der Hyper2000 alles betreibt und nur die Leistungsspitzen vom SF800 mit bedient werden.
Möchte nicht das der SF800 dauerhaft im Cluster gleich bedient wird wie der Hyper, da die dimensionierung der Akku's sonst gleich groß sein sollte, damit es sinnig ist, oder?

Automation II
= wenn P1 Sensor für 30sec unter -100W = aktiviere smartes Laden beim SF800 mit -P1-50W Wert
um eine Überstrom Verfolgung und laden zu erzeugen.

Könnte dies so funktionieren oder hab ich einen Denkfehler an einer stelle? Danke für Eure Gedanken & Hilfe.

harrymayr, 2025-07-10:
1. Ich bin nicht sicher, ob die Änderung der Cluster on the Fly geht, oder ob da ein Restart notwendig wird (hab nur einen Hyper).
2. Wird die Leistung entsprechend der Kapazität und Füllstand der Akkus verteilt. Beim gleichen Füllstand würde in deinem Fall der SF800 nur 1/5 beisteuern (wahrscheinlich noch weniger, wegen der 800W vs. 1200W). Ist aus meiner Sicht daher nicht notwendig die Cluster hin- und her zuschalten. Hyper auf `Cluster 2400W` und SF800 auf `Part of Hyper 2000 cluster`. Durch die maximale Leistung der beiden Devices kommt da maximal 2000W raus.

harrymayr, 2025-07-17:
Ich schließe das mal hier. Ideen und Ergänzungen passen dann besser in die Discussions.