Zendure/Zendure-HA #459: Does Local MQTT work using a ESP32 Bluetooth Proxy?

phlexss, 2025-07-11:
I find different info regarding using a ESP32 Bluetooth Proxy for Local Mqtt, but it's not clear if it should work or not.

So far, for me it doesn't seem to work.

I have added:

bluetooth_proxy:
  active: true
  connection_slots: 3

...to the configuration of the proxy in "ESPHome Device Builder" but that doesn't seem to make a difference.

Do I need to add other options in the proxy configuration?

Also, how does the Hyper2000 identify itself when I look in the bluetooth advertising monitor? I did see a "zen" device a few times but now I don't see this anymore and I have no idea which of the reported bluetooth devices could be the Hyper2000.

It would be great if I know if it can work or not using a ESP32 BT Proxsy, because going back to using the cloud-api after Local Mqtt fails is a pain in the ass everytime I need to do this....

Thanks for any info on the subject!

harrymayr, 2025-07-11:
I need to press the iot Button on the Hyper for 6 sec. to get rhe Hyper back to localmqtt once he was disconnected. And then I see him also in the BT advertisement list.

harrymayr, 2025-07-11:
No, I use the BT of my Raspberry. My hyper is shown as ZenE1_84, but I assume the 84 is depending on some serial numbers or so and will be different on your side.
The iot for around 6 second will reconnect to the WiFi (and also enables BT).

phlexss, 2025-07-11:
I wonder if anybody else has this working with an ESP32 BT Proxy....because otherwise this is a dead-end for me and I'll need to try seomthing else....

ralle6677, 2025-07-12:
> I wonder if anybody else has this working with an ESP32 BT Proxy....because otherwise this is a dead-end for me and I'll need to try something else....

I use two BT Proxy and it works for me.

Milvus11, 2025-07-13:
Hyper 2000 here and also using the ESP, works for me!

sky-future, 2025-07-21:
Look in your Bluetooth devices if you have a ai-thinker instead of Zen-XX that can be part of the issue! The code searches for Zen but in some cases it can be ai-thinker... 

[image]

bsmout, 2025-07-23:
Yes, I see ai-thinker as well. How can we fix this ?

harrymayr, 2025-07-24:
> Yes, I see ai-thinker as well. How can we fix this ?

1.1.0-pre7 does not support localMqtt for now, but in 1.1.0-pre3 you can replace line 252 in zenduremanager.py from 
`if si.name.startswith("Zen") and any(device.snNumber.endswith(e.decode("utf8")[:-1]) for e in si.manufacturer_data.values()):`
to 
`if si.address.startswith("94:C9:60") and any(device.snNumber.endswith(e.decode("utf8")[:-1]) for e in si.manufacturer_data.values()):`

have not tested it, but should work.

FireSon, 2025-07-24:
@bsmout In the new version this will be fixed, and the device recognision will also find aithinker devices. At this moment ble support is disabled, due to restructuring. Will add it this weekend, if all goes well 😄

bsmout, 2025-07-25:
Thanks. BTW is there a reason the file is called zendureRmanager.py ? ;)

bsmout, 2025-07-28:
Exactly the same problems here. I'm using a ESP32 Bluetooth proxy which sees both Hypers but I can't get it working. Status stays on BLE.

sky-future, 2025-07-31:
Tray version pre11 should work.

closed 2025-07-31 by FireSon