Zendure/Zendure-HA #464: Feature request

Zanooon97, 2025-07-14:
Hi,
I tried the smart power control and it's actually working better than the HEMS hehe, so I'm even more excited for when the local connection with the SF800 finally works.
My question is:
My SF800 1 charges faster than the second one. When the first one is full, it should cover the entire power demand and feed any excess energy into the household grid. The second unit should charge via PV and receive any additional energy via AC from the first unit.
This doesn't work yet.
Is it possible to set it up like that?

Best regards,
David

Hallo, ich habe die smarte leistungregelung probiert und aie klappt echt besser als der hems hehe deswegen bin ich um so froher wenn endlich die lokale verbindung mit dem SF 800 geht.

Meine frage ist:
Mein SF800 1 ist schneller aufgeladen als der zweite, wenn der erste foll ist soll der erste den gesamten strombedarf decken und den überschüssigen strom ins hausnetz abgeben, der zweite soll durch pv laden und den überschüssigen via ac von ersten.

das funktioniert noch nicht.

kann man das machen?

MFG David

FireSon, 2025-07-14:
Hi David, yes the local communication is being fixed. For the SF800/Pro/2400 the commands are send with the zenSDK, which basically is an HTTP API, and super fast 😄 

Your setup should be possible, also in the current situation, because you can set the bypass on the SF800 to always-on. So when the first one is full, the bypass will send the rest of the power to your house (and second SF800).

Zanooon97, 2025-07-15:
thats the setting i missing, i have seen it on screenshots of the hyper but I am missing this setting

[image]

harrymayr, 2025-07-15:
Copy the line with gridReverse from the hyper2000.py in the subfolder devices into solarflow800pro.py and restart HA.

[image]

Zanooon97, 2025-07-15:
It is not working harry, can you add this in the next update maybe?
In the app this function is showen as, export ecces energy or not.

I think its the bypass.

Zanooon97, 2025-07-15:
I have tried it now better, its working when I set it to Always On it pushes the hole energy in the house and the second one is charging the ecess enery!
But in the Auto mode is changing imediately to Always off?!
Is that normal?
Need I to set it manualy when the battery is full to Always On (Automation)?

Best regeards.

Zanooon97, 2025-07-16:
do you need a solarflow 800 for testing, I can send you if you live in europe. @FireSon Or @harrymayr

harrymayr, 2025-07-16:
That's nice 🙂, but for that I also need an (accesible) solarpanel. And the real developer is @FireSon, I just propose slight changes for the code. So from my side: No, thanks

Zanooon97, 2025-07-16:
I got it, calling http://192.168.1.33/properties/report in the browser gives me all data from my solarflow 800, now with post methods i can set settings. Great and easy way i think

Zanooon97, 2025-07-16:
harry can you say me witch file from the integration handles the local communication so i can try to edit it for me @harrymayr