Zendure/Zendure-HA #467: Question: Is the integration read-only or can it also send commands to Hyper 2000?

topfield007, 2025-07-16:
Hi FireSon,

first of all, thank you for your great work on the Zendure-HA integration!

I have one question:
Does the integration only read values (e.g. SoC, input/output, PV data, etc.), 
or is it also able to send commands back to the device (e.g. change modes, set charge/discharge limits)?

I could not find this explicitly in the README.

Thank you very much in advance!

Best regards 
Martin

topfield007, 2025-07-16:
Hello Maggi,

Changing the AC operating mode (AC Betriebsmodus) or turning the LED on/off does not take effect on the device.
The AC firmware is unknown, and the battery status is also shown as unknown.

[image]

Best regards
Martin

Maggi0r, 2025-07-16:
Okay, which integration version are you using? I'm using 1.1.0pre3. But I've had this problem a few times, too. Then I simply removed everything from the integration and re-downloaded it via HACS. And then set it up again. As you can see in the picture, that's what's displayed for me. I think that's why you can't control anything, since there doesn't seem to be a proper connection. Are you using local MQTT mode or purely cloud communication? I still have the cloud enabled.

[image]

Regards, Maggi

topfield007, 2025-07-17:
Now it works, thanks to your help.

Here’s what I did:

    Updated to the latest version 1.1.0pre3
    Disabled MQTT

Liebe Grüße / Best Regards Martin