Zendure/Zendure-HA #472: Requested Extension of Configuration Options…

fis-ne, 2025-07-18:
I’ve been using the Zendure Home Assistant integration for months now and I’m truly impressed!
There’s just one more setting I would love to see implemented, so I could completely do without the Zendure app:
I’d like to be able to select — or at least disable — the “Hyper Energy Program” on the Hyper2000, in order to manually configure the input and output power.

Is there any plan or possibility to expand the integration in this direction?
That would be amazing!

FireSon, 2025-07-22:
@fis-ne Not completely sure what you mean. 

If you set the Zendure Manager Operation to 'Off', it stops the smart mode, and you should be able to set the input/output limits to a value. So what am I missing that you would like t do? 
No problem to add some additional function if that would help you 😄

fis-ne, 2025-07-23:
Dear FireSon,
Thank you very much for your response.
Let me try to explain my concern:

I am using the integration with the following devices:
	•	1 x HYPER 200 (with 4 AB2000) in SmartCT mode, aligned with a ShellyPro3EM
	•	1 x HUB 2000 (with 3 AB2000) in manual mode, which supplies a base load of 60 to 250W via Home Assistant automations based on the delta between the SOC of the Hyper and the SOC of the HUB.

The Zendure Manager is always set to “off” in my case, as I do not use it.

I control all Zendure devices via Home Assistant using your HA integration (SOC-min, SOC-max, Bypass mode, various automations calculate, for example, when Hyper and HUB were last at 100% to then adjust SOC-min, I also set the output limit of the HUB via automations, etc.)

However, I occasionally want to charge the batteries of the Hyper to 100% from the grid.

To do this, I must first disable SmartCT mode in the Zendure app, because otherwise I cannot set a manual input limit — neither in the Zendure app nor in your HA integration. While I can set it there, it is not applied as long as SmartCT mode is active.

Therefore, I would really like to have the ability to disable and re-enable SmartCT mode directly within the HA integration.

[image]

[image]

fis-ne, 2025-07-25:
Thanks again for your quick reply.

Why I’m not using the “Smart Mode” is actually easy to explain:
When I discovered your Zendure Home Assistant integration by chance, I was immediately excited and installed it right away.
At that time, I didn’t understand the purpose of the Zendure Manager at all, nor could I find any detailed explanation.
I assumed that the Manager was developed for the Zendure Zen-Link clusters, which my HUB2000 doesn’t support.
So I simply left it set to “OFF” and used various Home Assistant automations to ensure that, among other things, the SOC values of my HYPER and HUB2000 would stay balanced.

Only after your message yesterday did I ask ChatGPT to explain the great possibilities your Zendure Manager actually offers.

To be honest, the effort to uninstall your integration and reinstall it in Smart Mode just to test it now feels too much for me – especially since I’d have to rebuild my carefully designed dashboards from scratch.

With my request being more of a “nice-to-have,” I absolutely don’t want to cause you any extra work.

For the few occasions where I want to fully charge the Hyper 2000 batteries from the grid, I can of course continue to manually deactivate the SmartCT mode in the Zendure app, charge the batteries to 100%, and then manually reactivate the SmartCT mode.

Thank you very much for your effort, and again, my compliments for this excellent integration.

Is there any way I can hit a “buy me a coffee” button for you?

Best regards,
Frank

Zanooon97, 2025-07-25:
He want this function in HA

[image]

FireSon, 2025-07-31:
@fis-ne I have asked the question about the possibility to add/remove the device fromHEMS, but that is rather complicated. I fear this is not something I can fix any time soon...

closed 2025-07-31 by FireSon