Zendure/Zendure-HA #479: Zendure HA stopped 1.1.0-pre3 working in the morning

bmoshous, 2025-07-22:
Hello,

since today in the morning the integration 1.1.0-pre3 is not working anymore. This was the only version working for me with SolarFlow 800 Pro.

I get the information "Diese Entität wird nicht mehr von der „zendure_ha"-Integration bereitgestellt. Wenn die Entität nicht mehr verwendet wird, lösche sie in den Einstellungen." for the entities.

To exclude config errors I restored the complete vm to multiple times in the past where it was working fine and checked if it is connecting again (no positive result).
I also tried different HA versions. The only one where I get at least no error connecting is the latest beta: Version 1.1.0-pre5

Nevertheless all sensors are "not available"
Sensoren
acStatus Nicht verfügbar
aiState Nicht verfügbar 
Akku entladen Nicht verfügbar
Akku geladen Nicht verfügbar

Please let me know which logs to provide to help finding the issue.

Thanks,
Basil

harrymayr, 2025-07-23:
Please try out the new version 1.1.0-pre7. If this is still not working, you need to enable debug logging.

harrymayr, 2025-07-24:
There are changes in the pre8 wrt ip adress. Can you please test this version?

bmoshous, 2025-07-24:
Hello,
the integration works again - all sensors I normally use are populated again!
Thanks for you support !

closed 2025-07-24 by bmoshous