Zendure/Zendure-HA #488: Hyper2000 localMqtt bad iot Message

FrColin, 2025-07-26:
I try to use the localMqtt via Ha addon with the last release 1.1.0-pr8

Message are received from hyper2000
but message send to the hyper use device name instead of devices key

here is some log
 
2025-07-27 00:46:48.892 INFO (paho-mqtt-client-56519:ef6f4f2e-68d1-441d-9152-122d13a55dc2) [custom_components.zendure_ha.api] Topic: iot/gDa3tb/Hyper2000/properties/read => {'properties': ['getAll']}

should be iot/gDa3tb/${device key}/

FrColin, 2025-07-27:
I found the reason in the code 
Why this replace ?
 _LOGGER.info(f"Topic: {self.name} {msg.topic.replace(deviceId, device.name)} => {payload}")
Better to not modify because it can cause misunderstandings

FireSon, 2025-07-28:
This is done to prevent exposing the deviceId by accendent when people share their logging