Zendure/Zendure-HA #500: 1.1.0-pre10

itapower, 2025-07-30:
Hi fireson,
at first, thank you very much for this great work.
I have installed the latest pre version, but I'm getting an error message, when I open the zendure integration.
Another question: is in this version bluetooth enable again? I want to use an esp bluetooth proxy for local mqtt.

[image]

FireSon, 2025-07-30:
@itapower Ai that the server gets into trouble is new for me, I mostly do that myself 😉 When does this message appear? Could you send a log file containing the debug logging when it happens to j.j.fireson at gamil.com ?

Yes it is, you can now also see the bluetooth mac addres in the device page

[image]

In this version, also if the Zen name is replaced into AIThinker, it should work. The bluetooth mac address for the devices is stored in home assistant.

itapower, 2025-07-30:
@FireSon thank for the quick reply. 
The message appears, when I try to add the integration.
I have send you a protocol to your email address. Hoping this is the right one.

FireSon, 2025-07-30:
Thanks I will look at it!

FireSon, 2025-07-30:
Just released pre11 which should fix this issue

SpeedyCarv, 2025-07-30:
> That has fixed it, thanks! I have another issue though, for some reason they are both stuck on discharging. The mode selected in the manager is not respected.
> 
> Restarted multiple times, switched the modes,... They both keep discharging at 1200W.
> 
> 2 hyper 2000's through the cloud mqtt.

Same here.

pulsarz, 2025-07-30:
Seems like p1 updates are not triggering. I set manual power mode and power to 100W andd it applies it. I switch to smart mode and it just stays at the previous output values.

pulsarz, 2025-07-30:
I can confirm I see no p1 udates in the logs like I used to see in pre3.

pulsarz, 2025-07-30:
Yeah, same entity I used before. It is a entity that gets updated through node-red though.
Just switched to the HW P1 entity but same thing, even restarted HA. Really strange.

Latest HA update.

pulsarz, 2025-07-30:
Switched back to pre3 and smart matching works fine (although only 1 of my 2 hypers since 3.0 wasn't detected back then).

SpeedyCarv, 2025-07-30:
Problem here is that nothing i change in the Zendure Manager has any effect. 

[image]

Only updates i make here

[image]

change the behavior of the Hyper.

1x Hyper 2000, Shelly 3EM, mqqt Mosquitto Add-on

MacdiverOne, 2025-07-30:
Maybe wrong reference assigned in steering program (as mentioned in #503)? Could happen since there were massive changes in last (pre-)releases. That's what a beta is for (...).

FireSon, 2025-07-30:
Would one of you be able to replace the code in manager.py to test this? You have to restart HA after you modify it...

```

    @callback
    async def _p1_changed(self, event: Event[EventStateChangedData]) -> None:
        try:
            # exit if there is nothing to do
            if not self.hass.is_running or not self.hass.is_running or (new_state := event.data["new_state"]) is None or self.operation == SmartMode.NONE:
                return

            try:  # convert the state to a float
                p1 = int(float(new_state.state))
            except ValueError:
                return

            # calculate the standard deviation
            avg = sum(self.zorder) / len(self.zorder) if len(self.zorder) > 1 else 0
            stddev = min(50, sqrt(sum([pow(i - avg, 2) for i in self.zorder]) / len(self.zorder)))
            if isFast := abs(p1 - avg) > SmartMode.Threshold * stddev:
                self.zorder.clear()
            self.zorder.append(p1)

            # check minimal time between updates
            time = datetime.now()
            if time < self.zero_next or (time < self.zero_fast and not isFast):
                return

            # get the current power
            powerActual = 0
            for d in self.devices:
                d.powerAct = await d.power_get()
                powerActual += d.powerAct

            _LOGGER.info(f"Update p1: {p1} power: {powerActual} operation: {self.operation} delta:{p1 - avg} stddev: {stddev} fast: {isFast}")
            match self.operation:
                case SmartMode.MATCHING:
                    if powerActual < 0:  # update when we are charging
                        self.update_power(min(0, powerActual + p1), ManagerState.CHARGING)
                    elif powerActual > 0:  # update when we are discharging
                        self.update_power(max(0, powerActual + p1), ManagerState.DISCHARGING)
                    elif self.zero_idle == datetime.max:  # check if it is the first time we are idle
                        _LOGGER.info(f"Wait {SmartMode.TIMEIDLE} sec for state change p1: {p1}")
                        self.zero_idle = time + timedelta(seconds=SmartMode.TIMEIDLE)
                    elif self.zero_idle < time:  # update when we are idle for more than SmartMode.TIMEIDLE seconds
                        if p1 < -SmartMode.MIN_POWER:
                            _LOGGER.info(f"Start charging with p1: {p1}")
                            self.update_power(p1, ManagerState.CHARGING)
                            self.zero_idle = datetime.max
                        elif p1 >= 0:
                            _LOGGER.info(f"Start discharging with p1: {p1}")
                            self.update_power(p1, ManagerState.DISCHARGING)
                            self.zero_idle = datetime.max
                        else:
                            _LOGGER.info(f"Unable to charge/discharge p1: {p1}")

                case SmartMode.MATCHING_DISCHARGE:
                    self.update_power(max(0, powerActual + p1), ManagerState.DISCHARGING)

                case SmartMode.MATCHING_CHARGE:
                    pwr = powerActual + p1 if powerActual < 0 else p1 if p1 < -SmartMode.MIN_POWER else 0
                    self.update_power(min(0, pwr), ManagerState.CHARGING)

                case SmartMode.MANUAL:
                    self.update_power(self.setpoint, ManagerState.DISCHARGING if self.setpoint >= 0 else ManagerState.CHARGING)

            self.zero_next = time + timedelta(seconds=SmartMode.TIMEZERO)
            self.zero_fast = time + timedelta(seconds=SmartMode.TIMEFAST)

        except Exception as err:
            _LOGGER.error(err)
            _LOGGER.error(traceback.format_exc())
```

SpeedyCarv, 2025-07-30:
I will try, just a moment.....

pulsarz, 2025-07-30:
Doesn't seem to fix it (I left _p1_changed1 intact).

SpeedyCarv, 2025-07-30:
same here. Still no smart matching.

SpeedyCarv, 2025-07-30:
That did the thing! Working now! Thank you so much!

pulsarz, 2025-07-30:
I can confirm it works now!

FireSon, 2025-07-30:
fixed in 1.1.0-pre12

closed 2025-07-30 by FireSon