Zendure/Zendure-HA #507: Smartmatching stops working for houers at night

Zanooon97, 2025-07-31:
Hi, i noticed that every night the outputpower goes to 0 for a couple of houers
maybe because the powerconsumption of the house is to low? when the inverter shuts off the powerconsumtion is around 60-70 watts.
can you check this?

[image]

harrymayr, 2025-07-31:
Can you add in this graph your P1 Sensor, so that it's clear how high the house demand is?

Zanooon97, 2025-07-31:
[image]

harrymayr, 2025-07-31:
You can give it a try, to reduce  MIN_POWER to 30 or 25 and START_POWER to 60 in const.py, then restart HA.

[image]

Zanooon97, 2025-08-01:
i solved it by reducing the start power and min power
but i try the new version now

Zanooon97, 2025-08-01:
top is working